natario1/CameraView #926: How to get the frames with watermark?

autismbug, 2020-07-28:
How to get frames with watermark?
I need to get the image frame with watermark and then call the push stream SDK to send the frame data.
Through the wiki I found that cameraview can only watermark previews, photos and recordings.

` 
cameraView.addFrameProcessor(new FrameProcessor() {
            @Override
            @WorkerThread
            public void process(@NonNull Frame frame) { }
 });
`
Frame data in this event does not contain watermarks.Is there any parameter setting that can meet my needs?
thanks

natario1, 2020-07-29:
You can use takePictureSnapshot() and use the picture data (JPEG)

autismbug, 2020-07-30:
25-30 frames a second .Call this method twenty times one second?

natario1, 2020-07-31:
I'd expect some 10-20 fps depending on device and light conditions, but you can see try and see how much you get.

closed 2020-08-29 by stale[bot]