natario1/CameraView #927: OrientationHelper.enable() called twice

thijsonline, 2020-07-29:
### Describe the bug
In CameraView.java, mOrientationHelper.enable() is called in onAttachedToWindow() AND open()
However, mOrientationHelper.disable() is ONLY called in onDetachedFromWindow()
Because two listeners will be registered and only one unregistered, when using the camera a lot, this could lead to the following error:
`Fatal Exception: java.lang.IllegalStateException: register failed, the sensor listeners size has exceeded the maximum limit 128
       at android.hardware.SystemSensorManager.registerListenerImpl(SystemSensorManager.java:197)
       at android.hardware.SensorManager.registerListener(SensorManager.java:854)
       at android.hardware.SensorManager.registerListener(SensorManager.java:761)
       at android.view.OrientationEventListener.enable(OrientationEventListener.java:93)
       at com.otaliastudios.cameraview.internal.utils.OrientationHelper.enable(OrientationHelper.java:106)`

- CameraView version: 2.6.3
- Camera engine used: camera2
- Reproducible in official demo app: unknown
- Device / Android version: *Nexus 5X, API 29*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
1. Add a breakpoint in OrientationHelper.enable() and OrientationHelper.disable()
2. Start the camera and close it
3. enable() will hit multiple times, disable() only once

### Expected behavior
Every call to enable() will eventually lead to a call to disable(), so that listeners won't stay in memory.

### XML layout
```xml
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:keepScreenOn="true"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        app:cameraGesturePinch="zoom"
        android:layout_marginRight="100dp"
        app:cameraGestureTap="autoFocus"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
        app:cameraGestureLongTap="autoFocus"
        app:cameraGestureScrollHorizontal="zoom"
        app:cameraGestureScrollVertical="exposureCorrection"
        app:cameraMode="picture"
        app:cameraAudio="off"/>
```

natario1, 2020-07-29:
Thanks! Would you like to fix it, since you have investigated? I think the helper could simply keep a boolean and avoid registering twice when enable() is called again.

thijsonline, 2020-07-30:
Done!
#928

closed 2020-08-02 by natario1