natario1/CameraView #935: Camera1 crash on API30

Asshes, 2020-08-07:
**Currently we are testing Android 11 Beta version [API 30]**

### Describe the bug
- CameraView version: *2.6.3*
- Camera engine used: *both*
- Reproducible in official demo app: *yes*
- Device / Android version: *Pixel 2XL/3XL, API 30*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Open Application
2. Press capture
3. See error

### Expected behavior
Expected that WHEN I click to take photo THEN app won't crash

### XML layout
<com.otaliastudios.cameraview.CameraView xmlns:app="http://schemas.android.com/apk/res-auto"
    android:id="@+id/camera"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    android:layout_gravity="center"
    android:keepScreenOn="true"
    app:cameraAudio="on"
    app:cameraCropOutput="false"
    app:cameraFacing="back"
    app:cameraFlash="auto"
    app:cameraGestureLongTap="none"
    app:cameraGesturePinch="zoom"
    app:cameraGestureScrollHorizontal="exposureCorrection"
    app:cameraGestureScrollVertical="zoom"
    app:cameraGestureTap="focusWithMarker"
    app:cameraGrid="off"
    app:cameraJpegQuality="100"
    app:cameraPlaySounds="true"
    app:cameraSessionType="picture" />

### Screenshots
No screenshots - it's just crash on taking photo action.

### Logs
**With Demo application**

2020-08-07 20:28:11.864 6336-6363/com.otaliastudios.cameraview.demo E/CameraEngine: CameraDevice.StateCallback reported an error: 4
2020-08-07 20:28:44.708 6336-6363/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: CameraViewEngine
    Process: com.otaliastudios.cameraview.demo, PID: 6336
    com.otaliastudios.cameraview.CameraException
        at com.otaliastudios.cameraview.engine.Camera2Engine$2.onError(Camera2Engine.java:453)
        at android.hardware.camera2.impl.CameraDeviceImpl.notifyError(CameraDeviceImpl.java:1629)
        at android.hardware.camera2.impl.CameraDeviceImpl.lambda$oDs27OTfKFfK18rUW2nQxxkPdV0(Unknown Source:0)
        at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$oDs27OTfKFfK18rUW2nQxxkPdV0.accept(Unknown Source:8)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:278)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:201)
        at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:97)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:223)
        at android.os.HandlerThread.run(HandlerThread.java:67)

----

2020-08-07 20:32:23.419 6681-6681/? I/cameraview.dem: Not late-enabling -Xcheck:jni (already on)
2020-08-07 20:32:23.433 6681-6681/? I/cameraview.dem: Unquickening 12 vdex files!
2020-08-07 20:32:23.434 6681-6681/? W/cameraview.dem: Unexpected CPU variant for X86 using defaults: x86
2020-08-07 20:32:23.714 6681-6681/com.otaliastudios.cameraview.demo W/ActivityThread: Application com.otaliastudios.cameraview.demo is waiting for the debugger on port 8100...
2020-08-07 20:32:23.731 6681-6681/com.otaliastudios.cameraview.demo I/System.out: Sending WAIT chunk
2020-08-07 20:32:24.536 6681-6681/com.otaliastudios.cameraview.demo I/System.out: Debugger has connected
2020-08-07 20:32:24.536 6681-6681/com.otaliastudios.cameraview.demo I/System.out: waiting for debugger to settle...
[... 89 lines cut ...]
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.RuntimeException: Error inside the onError callback. 2
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:272) 
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1268) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:223) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
2020-07-17 15:00:16.792 7123-7231/lt.sciil.spa E/CameraController: uncaughtException: Interrupting thread with state: STATE_STARTED due to CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 2
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 2
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:272)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1268)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:223)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.RuntimeException: Error inside the onError callback. 2
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:272) 
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1268) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:223) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 

### APK
We don't have github repository available for security reasons
https://play.google.com/store/apps/details?id=lt.sciil.eLPAAudit&hl=en 
Someone could connect me personally for login details: infernum.hircum@gmail.com

thijsonline, 2020-08-12:
Same here.
I think it is related to a bug in API 30: https://issuetracker.google.com/issues/158818461

Canato, 2020-09-08:
@natario1 would this be related to Permissions that changed in Android 11?

I believe this is the same issue as #799 

If is this the case, `Permissions + URI external issue`,  I could work on the fix for the camera.

Just need to know if this library need to write any data and if need to use Scoped Storage

natario1, 2020-09-08:
The library only writes on `File` and `FileDescriptor` that you pass to it, so you just have to make sure you pass files / descriptors that you have access to. It's not something to be fixed inside

Asshes, 2020-09-09:
We additionally found this, seems also related.
https://commonsware.com/blog/2020/08/16/action-image-capture-android-r.html

Canato, 2020-09-11:
The solution for us was change the URI from external to the authority(fileProvider) URI

val photoURI: Uri = FileProvider.getUriForFile(this, "com.example.android.fileprovider", it)
based on https://developer.android.com/training/camera/photobasics#kotlin

Hope this help more people

This change was based on the new Android 11 storage permissions https://developer.android.com/about/versions/11/privacy/storage
now we should not use WRITE/READ permissions but we can use a directly app storage

wataik, 2020-09-25:
I'm experiencing the same error when using emulator camera with VirtualScene mode while there is no issue with Emulated mode. The issue actually occurs with the demo app.

Here's the set up details.

- CameraView version: 2.6.4
- Camera engine used: both
- Reproducible in official demo app: yes
- Device / Android version: Android Studio Emulator (Pixel 2, API level 30)
- Android Studio version: 4.0

closed 2020-10-25 by stale[bot]