natario1/CameraView #940: CameraView crashing if using kotlin 1.4.0 and glsurface as preview rendered

filippodelfra, 2020-08-18:
### Describe the bug
If the project is set to use kotlin plugin 1.4.0 and glsurface as preview rendered the view will crash. Using texture everything works as exptected. 
The crash generated:
```
com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: GLThread 1048
    Process: com.otaliastudios.cameraview.demo, PID: 25268
    java.lang.NoSuchMethodError: No static method get-impl([II)I in class Lkotlin/UIntArray; or its super classes (declaration of 'kotlin.UIntArray' appears in /data/app/com.otaliastudios.cameraview.demo-CxUbH86-QP-L_J6AbBIO2Q==/base.apk)
        at com.otaliastudios.opengl.texture.GlTexture.<init>(GlTexture.kt:90)
        at com.otaliastudios.opengl.texture.GlTexture.<init>(GlTexture.kt:23)
        at com.otaliastudios.opengl.texture.GlTexture.<init>(GlTexture.kt:22)
        at com.otaliastudios.opengl.texture.GlTexture.<init>(Unknown Source:6)
        at com.otaliastudios.cameraview.internal.GlTextureDrawer.<init>(GlTextureDrawer.java:35)
        at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onSurfaceCreated(GlCameraPreview.java:148)
        at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1541)
        at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1272)
```
- CameraView version: 2.6.3
- Camera engine used: camera2
- Reproducible in official demo app: yes
- Device / Android version: OnePlus 3/9
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
Just set the demo project using kotlin 1.4.0:
```gradle
    extra["kotlinVersion"] = "1.4.0"
```
and cameraviewPreview to use glsurface:
```xml
    <com.otaliastudios.cameraview.CameraView
        ....
        app:cameraPreview="glSurface" />
```

sdeff, 2020-08-19:
I'm experiencing the same issue.

tobi512, 2020-08-19:
Just updated to Kotlin 1.4.0 and ran into the same problem!

FitApps7, 2020-08-24:
also for me, on google pixel emulator android 10

asozcan, 2020-08-24:
Same here, currently not able to upgrade to kotlin:1.4.0

reactivedroid, 2020-08-27:
We are also seeing this crash when you update `CameraView` to `2.6.3` with Kotlin 1.4. For now, moving to a lower version `2.6.2` works fine

commit 8207e676 (2020-09-02): Some bug fixes (#953)

* Kotlin 1.4.0, fixes #940

* Fix multifilter bug, fixes #875

* Fix arithmetic exception, fixes #895

* Update dependencies

* Demo app in Kotlin

* Avoid stackoverflows on zoom/ev, fixes #856

* Remove Gemfile.lock

* Try to fix build
closed 2020-09-02 by natario1