natario1/CameraView #948: Is it possible to set horizontal orientation to camera instead of depending on auto rotate?

BartlomiejFrankow, 2020-08-26:
Can't find nothing in documentation. I know this is possible to set `app:cameraUseDeviceOrientation="true"` but I want to set horizontal orientation and don't change it even when user will move device.

MayconCardoso, 2020-08-26:
Hey, why don't you set the orientation settings on your activity on your AndroidManifest? It doesn't look it is responsibility of this library. 

```xml
<activity android:name=".YourActivity"  
     android:screenOrientation="landscape"/>  
``` 

Or programatically if it is a fragment on yout `onCreate` method.

### Java 
```java
setRequestedOrientation(ActivityInfo.SCREEN_ORIENTATION_LANDSCAPE);
```

### Kotlin
```kotlin
requestedOrientation = ActivityInfo.SCREEN_ORIENTATION_LANDSCAPE
```

BartlomiejFrankow, 2020-08-26:
@MayconCardoso I do have this done. But camera orientation seems to not be connected with activity orientation but with device orientation so even if I do have `android:screenOrientation="landscape"` inside `AndroidManifest` camera will be vertical if I handling phone in vertical way. 

I know this is edge case but in my case it's pretty important. I'm playing video and recording user reaction with this camera. User before go to video is in vertical mode always but video player itself is in new activity (landscape) I'm loading this video in background so if user is doing this many times he can put device on flat surface (was in vertical view) and device will not trigger changed orientation to landscape and camera will still be vertical.

Just need to set camera in vertical way and lock.

closed 2020-09-27 by stale[bot]

Radiokot, 2020-10-07:
I have nearly the same issue. My capture activity orientation is locked to `portrait` and I have set  `app:cameraUseDeviceOrientation="false"`
It works fine in all cases but when the device is laying on a perfectly aligned table (0, 0, 0). In such case the result bitmap is landscape anyway, I think it takes natural camera orientation which is landscape and uses it to save the file