natario1/CameraView #951: how to get a right crop coordinates according to the preview bitmap?

Greatkingcj, 2020-09-01:
Hi,
i take a full screen picture and  place an Image View on top of CameraView, then i could crop this picture. but i found that  it will be difficult to get the coordinates with the real image. cause that coordinates are according to the screen and not according to the bitmap itself. （the size of bitmap and screen are not the same）
`<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:adjustViewBounds="true"
        android:keepScreenOn="true"
        app:cameraUseDeviceOrientation="false"
        app:cameraEngine="camera1"
        app:cameraPlaySounds="false"
        app:cameraAudio="off"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent"/>`
Any help or idea is highly appreciated.

xxxifan, 2020-09-06:
same here. 
I used `takePicture` instead to get correct size.

Greatkingcj, 2020-09-07:
> same here.
> I used `takePicture` instead to get correct size.

I have tried takePicture to get a full screen picture, but the size of bitmap are not same to screen, can you show me how you do this ? thanks a lot !!

xxxifan, 2020-09-07:
for 1080x1920
```
cameraView.setPreviewStreamSize { arrayListOf(Size(1080, 1920)) }
cameraView.setPictureSize { arrayListOf(Size(1080, 1920)) }
```

Greatkingcj, 2020-09-07:
> for 1080x1920
> 
> ```
> cameraView.setPreviewStreamSize { arrayListOf(Size(1080, 1920)) }
> cameraView.setPictureSize { arrayListOf(Size(1080, 1920)) }
> ```
> 
> then you can get full screen preview and actually 1080x1920 bitmap.

i tried this before, but not all devices are 1080 * 1920,  the preview size the camera support are not same as the screen size, so i can not set screen size to cameraView

closed 2020-10-10 by stale[bot]