natario1/CameraView #955: ANR While initialization camera.

FitApps7, 2020-09-03:
- CameraView version: *2.6.3*
- Camera engine used: *camera1*
- Reproducible in official demo app: *no*
- Device / Android version: *Samsung Galaxy J2 Core (j2corelte), 1024MB RAM, Android 8.1*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
Just opened activity in which camera code existed

### Expected behavior
Camera preview should be open.

### XML layout

```xml
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/view_cam"
        android:layout_width="match_parent"
        android:layout_height="0dp"
        android:foregroundGravity="center"
        android:keepScreenOn="true"
        app:cameraAudio="off"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
        app:cameraEngine="camera1"
        app:cameraExperimental="true"
        app:cameraFacing="back"
        app:cameraFlash="off"
        app:cameraGestureLongTap="none"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="none"
        app:cameraGestureScrollVertical="exposureCorrection"
        app:cameraGestureTap="autoFocus"
        app:cameraGrid="off"
        app:cameraMode="picture"
        app:cameraPictureSizeBiggest="true"
        app:cameraPlaySounds="false"
        app:cameraPreview="glSurface"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintTop_toTopOf="@+id/view_bg_top_bar_full_cam" />
```
This ANR occur on this device 
```
Samsung Galaxy J2 Core (j2corelte), 1024MB RAM, Android 8.1
Report 1 of 1
"main" tid=1 Native
"main" prio=5 tid=1 Native | group="main" sCount=1 dsCount=0 flags=1 obj=0x71e65f00 self=0xe9a2e000 | sysTid=2884 nice=-11 cgrp=default sched=0/0 handle=0xeda394a4 | state=S schedstat=( 43394396997 13691475908 122521 ) utm=3028 stm=1310 core=3 HZ=100 | stack=0xff687000-0xff689000 stackSize=8MB | held mutexes=
#00 pc 0000000000049498 /system/lib/libc.so (__ioctl+8)
#01 pc 000000000001de9d /system/lib/libc.so (ioctl+40)
#02 pc 0000000000042549 /system/lib/libbinder.so (android::IPCThreadState::talkWithDriver(bool)+204)
#03 pc 0000000000042f5f /system/lib/libbinder.so (android::IPCThreadState::waitForResponse(android::Parcel*, int*)+246)
#04 pc 000000000003d3dd /system/lib/libbinder.so (android::BpBinder::transact(unsigned int, android::Parcel const&, android::Parcel*, unsigned int)+36)
#05 pc 0000000000029269 /system/lib/libcamera_client.so (android::hardware::BpCameraService::getCameraInfo(int, android::hardware::CameraInfo*)+100)
#06 pc 0000000000037d89 /system/lib/libcamera_client.so (android::CameraBase<android::Camera, android::CameraTraits<android::Camera>>::getCameraInfo(int, android::hardware::CameraInfo*)+40)
#07 pc 00000000000de603 /system/lib/libandroid_runtime.so (???)
#08 pc 00000000001fcbe5 /system/framework/arm/boot-framework.oat (Java_android_hardware_Camera__1getCameraInfo__ILandroid_hardware_Camera_00024CameraInfo_2+116)
at android.hardware.Camera._getCameraInfo (Camera.java)
at android.hardware.Camera.getCameraInfo (Camera.java:260)
at com.otaliastudios.cameraview.CameraUtils.hasCameraFacing (CameraUtils.java:68)
at com.otaliastudios.cameraview.controls.Facing.DEFAULT (Facing.java:33)
at com.otaliastudios.cameraview.controls.ControlParser.<init> (ControlParser.java:32)
at com.otaliastudios.cameraview.CameraView.initialize (CameraView.java:189)
at com.otaliastudios.cameraview.CameraView.<init> (CameraView.java:176)
at java.lang.reflect.Constructor.newInstance0 (Constructor.java)
at java.lang.reflect.Constructor.newInstance (Constructor.java:334)
at android.view.LayoutInflater.createView (LayoutInflater.java:647)
at android.view.LayoutInflater.createViewFromTag (LayoutInflater.java:790)
at android.view.LayoutInflater.createViewFromTag (LayoutInflater.java:730)
at android.view.LayoutInflater.rInflate (LayoutInflater.java:863)
at android.view.LayoutInflater.rInflateChildren (LayoutInflater.java:824)
at android.view.LayoutInflater.inflate (LayoutInflater.java:515)
- locked <0x0e30ea3a> (a java.lang.Object[])
at android.view.LayoutInflater.inflate (LayoutInflater.java:423)
at android.view.LayoutInflater.inflate (LayoutInflater.java:374)
at androidx.appcompat.app.AppCompatDelegateImpl.setContentView (AppCompatDelegateImpl.java:696)
at androidx.appcompat.app.AppCompatActivity.setContentView (AppCompatActivity.java:170)
at com.hazel.cam.scanner.free.activity.camera.FullCaptureCameraActivity.onCreate (FullCaptureCameraActivity.java:62)//setContentView(R.layout.activity_full_capture_camera)
at android.app.Activity.performCreate (Activity.java:7009)
at android.app.Activity.performCreate (Activity.java:7000)
at android.app.Instrumentation.callActivityOnCreate (Instrumentation.java:1214)
at android.app.ActivityThread.performLaunchActivity (ActivityThread.java:2742)
at android.app.ActivityThread.handleLaunchActivity (ActivityThread.java:2872)
at android.app.ActivityThread.-wrap11 (ActivityThread.java)
at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1595)
at android.os.Handler.dispatchMessage (Handler.java:106)
at android.os.Looper.loop (Looper.java:164)
at android.app.ActivityThread.main (ActivityThread.java:6543)
at java.lang.reflect.Method.invoke (Method.java)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:440)
at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:810)
```

### APK
Link to a Github repo where the bug is reproducible.

Rajatsonii, 2020-09-14:
I have same issue

closed 2020-10-17 by stale[bot]

EzequielAdrianM, 2023-05-28:
By looking at the log, you can see that the code freeze at getCameraInfo() so the device is probably unable to find any cameras. CameraView is already handling camera operations on a background thread, so there is no reason for ANR unless unpredictable hardware issues like this one. Just like Natario mentioned, you should probably restart the phone.