natario1/CameraView #956: App crashes in some devices with Android version Q while creating the shader

SreeniMannem, 2020-09-05:
App is crashing when camera is opened for streaming and exiting the app.

Every time app crashes when open the camera start the streaming, it is happening in only particular devices like Redmi, Samsung, Oppo (Version Q) devices it seems.

- CameraView version: 2.6.3
- Camera engine used: camera2
- Reproducible in official demo app: yes
- Device / Android version: *Redmi Note8 pro, Oppo and Samsung, API 28
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Click on the app
2. Start video recording using camera2
3. App crashes while creating the opengl shaders
4. App exists

### Expected behavior
App should not crash and device should be able create a video stream

BaseFilter.Java

return "#extension GL_OES_EGL_image_external : require\n"
                + "precision mediump float;\n"
                + "varying vec2 "+fragmentTextureCoordinateName+";\n"
                + "uniform samplerExternalOES sTexture;\n"
                + "void main() {\n"
                **+ "  gl_FragColor = texture2D(sTexture, "+fragmentTextureCoordinateName+");\n"**
                + "}\n";

### Logs
09-04 16:23:47.594 22175 24568 E AndroidRuntime: FATAL EXCEPTION: GLThread 3392

09-04 16:23:47.594 22175 24568 E AndroidRuntime: Process: com.vsoftlabs.cameraview, PID: 22175

**09-04 16:23:47.594 22175 24568 E AndroidRuntime: java.lang.RuntimeException: Error during glCreateShader type=35632: glError 0x502: invalid operation**

09-04 16:23:47.594 22175 24568 E AndroidRuntime: at com.otaliastudios.opengl.core.Egloo.checkGlError(Egloo.kt:34)

09-04 16:23:47.594 22175 24568 E AndroidRuntime: at com.otaliastudios.opengl.program.GlProgram$Companion.createShader(GlProgram.kt:76)

09-04 16:23:47.594 22175 24568 E AndroidRuntime: at com.otaliastudios.opengl.program.GlProgram$Companion.create(GlProgram.kt:91)

09-04 16:23:47.594 22175 24568 E AndroidRuntime: at com.otaliastudios.opengl.program.GlProgram.create(Unknown Source:2)

09-04 16:23:47.594 22175 24568 E AndroidRuntime: at com.otaliastudios.cameraview.internal.GlTextureDrawer.draw(GlTextureDrawer.java:75)

09-04 16:23:47.594 22175 24568 E AndroidRuntime: at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame(GlCameraPreview.java:276)

09-04 16:23:47.594 22175 24568 E AndroidRuntime: at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1573)

09-04 16:23:47.594 22175 24568 E AndroidRuntime: at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1272)

closed 2020-10-03 by stale[bot]

PraveenOjha, 2020-10-14:
Possibility of liked reason  https://github.com/natario1/CameraView/issues/916#issue-657133584

PraveenOjha, 2020-10-14:
Please use an old built till this issue is solved , try 2.6.1

parekhpratik24, 2020-11-07:
Is this resolved? @PraveenOjha : there is no special code, it is just multifilter that crashes on some devices. 
Root cause is this line : AndroidRuntime: at com.otaliastudios.opengl.core.Egloo.checkGlError(Egloo.kt:34)