natario1/CameraView #962: Video - onVideoTaken event not called onPause

rai2270, 2020-09-19:
- Taking a video using camera.takeVideo().
- If the Fragment call camera.stopVideo(): 2 events are called: onVideoRecordingEnd  &  onVideoTaken. Everything works fine.
- But going to the background (onPause) while video is recording (inside Fragment) -> onVideoRecordingEnd event is called, but onVideoTaken is not.

I've put my UI changes inside onVideoRecordingEnd , and some logic related to file saving in onVideoTaken because only in that event we get VideoResult result.getFile().getAbsolutePath().

rai2270, 2020-09-21:
**In my code, this issue only happened for Fragment.**

I solved this in a strange way:
1) Comment out setLifecycleOwner
2) Add onResume, Pause, Destroy
3) Comment out cameraview.destroy()
		
    @Override
    public void onViewCreated(final View view, Bundle savedInstanceState) {
        super.onViewCreated(view, savedInstanceState);
        //camera.setLifecycleOwner(getViewLifecycleOwner());
		...
	}
		
    @Override
    public void onResume() {
        super.onResume();
        binding.cameraview.open();
    }

    @Override
    public void onPause() {
        super.onPause();
        if (binding.cameraview.isTakingVideo()) {
            binding.cameraview.stopVideo();
        }
        binding.cameraview.close();
    }

    @Override
    public void onDestroy() {
        super.onDestroy();
        //binding.cameraview.destroy();
    }

closed 2020-10-22 by stale[bot]

sathishmscict, 2021-01-25:
Video pause and continue recording not working