natario1/CameraView #963: image size

nawazish011, 2020-09-23:


nawazish011, 2020-09-23:
When image is capture it is not capture according to imageView size,  it capture image according to device resolution how we can capture image of custom width and height.,

mucahidkambur, 2020-09-24:
If you want to take photo with same height&width you need to use takeSnapshot, otherwise you need to use the SizeSelectors

https://natario1.github.io/CameraView/docs/capture-size

nawazish011, 2020-09-24:
i have already use size selector technique but in size selector it give no size

mucahidkambur, 2020-09-24:
Don't override it. Open CameraBaseEngine.java class and go line 849. That result will be your picture resolution. Also you can see supported size with "sizes" variable.

nawazish011, 2020-09-24:
In size selector multiple sizes returned can we select one specific size for picture. i.e 768X1024 which is available in size list

mucahidkambur, 2020-09-24:
Yes you can with cameraView.setPictureSize { arrayListOf(Size(768, 1024)) }. But i don't recommend to use hard coded resolutions because some devices may not support 768x1024.

nawazish011, 2020-09-24:
Thank you so much for your support. This solution solved my problem. I need only that specific resolution because i have to process image to extract data from it.

nawazish011, 2020-09-28:
Thanks for your reply, but the issue is I want to capture image on landscape i.e 1024X768 resolution keeping my phone on portrait orientation if phone is on portrait orientation the image the resolution is like 768X1024 but i want to get image resolution (1024X768) by keeping my phone on portrait orientation.

PavelLida, 2020-10-20:
may be you also can help me. How can i make the image always 4 by 3 (ratio)?
in my xml `custom:cameraPictureSizeAspectRatio="4:3"`
but this property is ignored by many devices and show (pictureSize) a 16 by 9

mucahidkambur, 2020-11-09:
> Thanks for your reply, but the issue is I want to capture image on landscape i.e 1024X768 resolution keeping my phone on portrait orientation if phone is on portrait orientation the image the [...]

When device rotated you have to set picture with landscape ratio(arrayListOf(Size(1024, 768))) then restart the camera.

> may be you also can help me. How can i make the image(pictureSize) always 4 by 3 (ratio)?
> in my xml i use `custom:cameraPictureSizeAspectRatio="4:3"`
> but this property is ignored by many devices and show result (pictureSize) a 16 by 9

```
val ratio = SizeSelectors.aspectRatio(AspectRatio.of(4, 3), 0f)
val sizeSelector = SizeSelectors.or(ratio)
cameraView.setPictureSize(sizeSelector)
```

closed 2020-12-12 by stale[bot]