natario1/CameraView #970: Crash No static method GlTexture

kelvinsaputra, 2020-10-12:
### Describe the bug

I've read the bug about kotlin 1.4 solved in 2.6.4. I believe this is the same bug, except it's not in kotlin 1.4, but 1.3.7.10.
When I tried to update the lib from 2.6.1 to 2.6.4, I can't open the camera. It crashed right away. Here's the stack trace I got from the crash.
java.lang.NoSuchMethodError: No static method get-pVg5ArA([II)I in class Lkotlin/UIntArray; or its super classes (declaration of 'kotlin.UIntArray' appears in base.apk!classes28.dex)
	at com.otaliastudios.opengl.texture.GlTexture.<init>(GlTexture.kt:89)
	at com.otaliastudios.opengl.texture.GlTexture.<init>(GlTexture.kt:21)
	at com.otaliastudios.opengl.texture.GlTexture.<init>(GlTexture.kt:20)
	at com.otaliastudios.opengl.texture.GlTexture.<init>(Unknown Source:6)
	at com.otaliastudios.cameraview.internal.GlTextureDrawer.<init>(GlTextureDrawer.java:35)
	at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onSurfaceCreated(GlCameraPreview.java:148)
	at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1539)
	at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1270)

- CameraView version: 2.6.4
- Camera engine used: camera2
- Reproducible in official demo app: no
- Device / Android version: Xiaomi Redmi Note 5
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
Open the camera, then crash

### Expected behavior
Not crash

### Screenshots
No Screenshot, it crashed.

Thank you.

anxiao521, 2020-10-21:
So did I,Have you solved it?

anxiao521, 2020-10-21:
com.otaliastudios:cameraview:2.6.2 Is no problem

kelvinsaputra, 2020-10-23:
So far no luck

closed 2020-11-25 by stale[bot]