natario1/CameraView #971: java.lang.RuntimeException

cuibg, 2020-10-13:
### Describe the bug
When I use camera1 in my application，the error has always appeared！So I think it should be resolved !
1. Exception:
java.lang.RuntimeException
setParameters failed
2. description
this bug has appeared 53 times!
[image]
[image]

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to takepicture
2. Click on open flash
3. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### XML layout

```xml
<CameraView
            android:id="@+id/lbj_camera_view"
            android:layout_width="0dp"
            android:layout_height="0dp"
            android:keepScreenOn="true"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toTopOf="parent"
            camera:cameraAudio="off"
            camera:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
            camera:cameraEngine="camera1"
            camera:cameraExperimental="false"
            camera:cameraGesturePinch="zoom"
            camera:cameraGestureScrollHorizontal="filterControl1"
            camera:cameraGestureScrollVertical="exposureCorrection"
            camera:cameraGestureTap="autoFocus"
            camera:cameraHdr="off"
            camera:cameraMode="picture"
            camera:cameraPictureSizeMaxWidth = "@integer/picture_max_width"
            camera:cameraPictureFormat="jpeg"
            camera:cameraPictureMetering="true"
            camera:cameraPictureSizeAspectRatio="@string/picture_size_aspect_ration"
            camera:cameraPictureSizeBiggest="true"
            camera:cameraPlaySounds="false"
            camera:cameraPreview="glSurface"
            camera:cameraPreviewFrameRateExact="true"
            camera:cameraRequestPermissions="true"
            camera:cameraUseDeviceOrientation="false"
            camera:cameraWhiteBalance="auto" />
```

cuibg, 2020-10-13:
@natario1

cuibg, 2020-10-13:
[image]
I believe that try{}catch{}this exception can resolve this question！

cuibg, 2020-10-16:
up

73k05, 2020-10-16:
Hi coders,

I've got this crash as well, here is the full stack
```
Fatal Exception: java.lang.RuntimeException
setParameters failed
Fatal Exception: java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:2117)
       at com.otaliastudios.cameraview.engine.Camera1Engine$11.run(Camera1Engine.java:885)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(zzi.java:10)
       at com.google.android.gms.tasks.zzq.zza(zzq.java:19)
       at com.google.android.gms.tasks.zzu.zzb(zzu.java:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(zzi.java:10)
       at com.google.android.gms.tasks.zzq.zza(zzq.java:19)
       at com.google.android.gms.tasks.zzu.zzb(zzu.java:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(zzi.java:10)
       at com.google.android.gms.tasks.zzq.zza(zzq.java:19)
       at com.google.android.gms.tasks.zzu.zzb(zzu.java:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(zzi.java:10)
       at com.google.android.gms.tasks.zzq.zza(zzq.java:19)
       at com.google.android.gms.tasks.zzu.zzb(zzu.java:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
[... cut ...]
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1152)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
       at java.lang.Thread.run(Thread.java:919)
queued-work-looper
queued-work-looper
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:336)
       at android.os.Looper.loop(Looper.java:197)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```

cuibg, 2020-10-19:
@73k05 I can't solve this problem！Do you have any other ideas？

73k05, 2020-10-19:
> @73k05 I can't solve this problem！Do you have any other ideas？

Nop, I can't reproduce myself, did you try with engine 2?

cuibg, 2020-10-20:
@73k05 ，Engine 2 has many bugs in some devices！

cuibg, 2020-10-26:
up

cuibg, 2020-10-28:
up

cuibg, 2020-10-30:
@73k05 @shehabic-work go on up

cuibg, 2020-11-02:
@natario1 hi,I hope it can be resolved！thanks

cuibg, 2020-11-03:
hi，the author is very clever！When project has runtime exception,it will be throwed.
[image]

closed 2020-11-03 by cuibg