natario1/CameraView #972: RuntimeException StackOverflowError CameraOrchestrator onComplete

73k05, 2020-10-13:
### Describe the bug
- CameraView version: *com.otaliastudios:cameraview:2.6.4*
- Camera engine used: *engine 2*
- Reproducible in official demo app: *did not try*
- Device / Android version: *Android 10 - Galaxy S10*

### To Reproduce
No idea, coming from community through crashlytics

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera_view"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:keepScreenOn = "true"
        android:layout_gravity="center"
        app:cameraFacing = "back"
        app:cameraFlash = "off"
        app:cameraGestureLongTap = "autoFocus"
        app:cameraGesturePinch = "zoom"
        app:cameraGestureTap = "autoFocus"
        app:cameraPictureFormat = "jpeg"
        app:cameraVideoCodec = "deviceDefault"
        app:cameraWhiteBalance = "auto"
        />
```

### Screenshots
[image]

### Logs
```

Fatal Exception: java.lang.RuntimeException
java.lang.StackOverflowError: stack size 1039KB
Fatal Exception: java.lang.RuntimeException: java.lang.StackOverflowError: stack size 1039KB
       at com.otaliastudios.cameraview.engine.CameraEngine$2.run(CameraEngine.java:240)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:237)
       at android.app.ActivityThread.main(ActivityThread.java:8167)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
Caused by java.lang.StackOverflowError
stack size 1039KB
Caused by java.lang.StackOverflowError: stack size 1039KB
       at com.google.android.gms.tasks.zzu.getException(zzu.java:27)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:104)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(zzi.java:10)
       at com.google.android.gms.tasks.zzq.zza(zzq.java:19)
       at com.google.android.gms.tasks.zzu.zzb(zzu.java:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
[... 311 lines cut ...]
       at java.lang.Thread.run(Thread.java:919)
glide-disk-cache-thread-0
glide-disk-cache-thread-0
       at sun.misc.Unsafe.park(Unsafe.java)
       at java.util.concurrent.locks.LockSupport.park(LockSupport.java:190)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2067)
       at java.util.concurrent.PriorityBlockingQueue.take(PriorityBlockingQueue.java:548)
       at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1092)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1152)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
       at java.lang.Thread.run(Thread.java:919)
       at com.bumptech.glide.load.engine.executor.GlideExecutor$DefaultThreadFactory$1.run(GlideExecutor.java:393)
```
### APK
https://play.google.com/store/apps/details?id=com.pixstar.snap

### Duplicate
I have a similar crash on 
```
CameraView.java line 2264
com.otaliastudios.cameraview.CameraView$CameraCallbacks.dispatchOnFocusStart
...
com.otaliastudios.cameraview.engine.CameraEngine$2.run (CameraEngine.java:240)
```
Should I post the stak here as well or create a new ticket?

cuibg, 2020-10-26:
@73k05 hi，are there many bugs in engine 2？

cuibg, 2020-11-04:
this error is belong to camera！

closed 2020-12-12 by stale[bot]