natario1/CameraView #973: How to check camera support flash or not?

24Lathiya, 2020-10-14:
### How do I?
I would like hide flash icon if camera not support flash. for example many front cameras does not support flash.

### Version used
CameraView exact version : 2.6.4

PraveenOjha, 2020-10-25:
CameraOptions cameraOptions=camView.getCameraOptions();
Collection modes  cameraOptions.getSupportedFlash();
boolean b= modes.contains(Flash.ON);

can check for Flash.torchetc

24Lathiya, 2020-10-27:
@PraveenOjha getting `mCameraView.getCameraOptions() ` null at initialise camera
@natario1 Please let me know if any solution.

24Lathiya, 2020-10-28:
Thanks @PraveenOjha your trick work like charm with few changes like below

```
new Handler().postDelayed(new Runnable() {
            @Override
            public void run() {
                CameraOptions cameraOptions = mCameraView.getCameraOptions();
                if (cameraOptions != null) {
                    Collection<Flash> modes = cameraOptions.getSupportedFlash();
                    boolean isSupportFlash = modes.contains(Flash.ON);
                    System.out.println("=====flash=====" + isSupportFlash);
                }
            }
        },300);
```

closed 2020-10-28 by 24Lathiya