natario1/CameraView #975: Select preview size with no crops

ValYouW, 2020-10-18:
### How do I?
Hi, how can I make sure that cameraview will choose a preview resolution that matches the device's native aspect ratio?
In other words, how can I make sure I get the "most zoomed-out" picture available?

My specific case:
On Nexus 5X, **portrait** activity, I setup cameraview component to match the entire screen:
```
<com.otaliastudios.cameraview.CameraView
	android:id="@+id/camera_view"
	android:layout_width="wrap_content"
	android:layout_height="wrap_content"
	android:layout_gravity="center"
	app:layout_constraintEnd_toEndOf="parent"
	app:layout_constraintTop_toTopOf="parent"
	app:layout_constraintStart_toStartOf="parent">
</com.otaliastudios.cameraview.CameraView>
```

The selected preview size is 1920*1080 (16:9), if the selected preview was any of the 4:3 aspect ratios the camera would have been more "zoomed-out" (yes, I would get black margins but thtat's ok). I don't want any crop to happen.

Thx!

### Version used
2.6.4

natario1, 2020-10-18:
Nice question, I have never thought about it from this point of view. One way is to use     `app:cameraPictureSizeAspectRatio="4:3"`, it should work fine for the vast majority of devices.

Another is to use `cameraView.setPreviewStreamSize(SizeSelector)` and choose the size yourself.

ValYouW, 2020-10-18:
I would choose myself if I knew what's the device's camera native aspect ratio, do you know of any Android API to get that info?
Currently I have some logic for selection which I'm not sure is correct...

ValYouW, 2020-10-19:
Just saw `getPictureSize` which seems to return the "correct" aspect ratio... is this info coming from the OS?

PavelLida, 2020-10-20:
I have the same problem. I want to make the capture and snapshot the same (4 : 3).

I use in xml property
`custom: cameraPictureSizeAspectRatio = "4: 3"`

it works on many devices. But as practice shows, not at all (in particular, on Xiaomi MI A1) the picture turns out to be 16 by 9, although 4 by 3 is needed. And the most interesting thing is that MI A1 supports 4 by 3, but shows me it to 16 by 9

natario1, 2020-10-20:
The picture/video size is chosen among the available sizes, by default choosing the biggest one (unless you use setPictureSize/setVideoSize). I would expect that in 99% of cases, the biggest size matches the sensor aspect ratio, unless some vendor has a big sensor but can't support capturing on it with full resolution.

Once picture/video size is selected, the preview stream size is automatically selected to match the aspect ratio of the previous step (*unless you call setPreviewStreamSize*). So in theory you should just do nothing and it already does what you want, as long as the view is wrap_content with no weird constraints

ValYouW, 2020-10-22:
Doesn't work in my case...
Debugging `CameraBaseEngine::computePreviewStreamSize` I get:
```
targetRatio = "3:4"
targetMinSize = "1080x1731"

sizes = [
  0 = {Size@5704} "1080x1920"
  1 = {Size@5705} "1200x1600"
  2 = {Size@5706} "1080x1440"
  3 = {Size@5707} "960x1280"
  4 = {Size@5708} "768x1280"
  5 = {Size@5709} "720x1280"
  6 = {Size@5710} "768x1024"
  7 = {Size@5711} "600x800"
  8 = {Size@5712} "480x864"
  9 = {Size@5713} "480x800"
  10 = {Size@5714} "480x720"
  11 = {Size@5715} "480x640"
  12 = {Size@5716} "360x640"
  13 = {Size@5717} "288x352"
  14 = {Size@5718} "240x320"
  15 = {Size@5719} "144x176"
  16 = {Size@5720} "120x160"
]

result = "1080x1920"
```
The selected result ratio is "9:16", I would expect the result to be "1200x1600" or better "1080x1440"...

ValYouW, 2020-11-14:
@natario1 don't you think the above is a bug?

closed 2020-12-24 by stale[bot]