natario1/CameraView #978: Crash on camera recording on Xiaomi Redmi 8, Android version 9

Radcat501, 2020-10-20:
Hi! 

I implemented a fragment, which recording video, pack data to MultipartBody and send it to server. This workes fine on Xiaomi Mi Play with Android 8, but on Xiaomi Redmi 8 with Android 9 i faced a crash, and i still cant find solution. Here is stacktrace^

### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *version number*
- Camera engine used: *camera1/camera2/both*
- Reproducible in official demo app: *yes/no*
- Device / Android version: *Pixel, API 28*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes/no*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to '...'
2. Click on '...'
3. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<CameraView
    android:layout_width="match_parent"
    android:layout_height="match_parent"       
    ...>
</CameraView>
```

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
Use `CameraLogger.setLogLevel(LEVEL_INFO)` to see all logs into LogCat.

Use `CameraLogger.registerLogger()` to export to file or crash reporting service.

### APK
Link to a Github repo where the bug is reproducible.

Radcat501, 2020-10-20:
Also, i am using api ("com.otaliastudios:cameraview:2.6.4")

PraveenOjha, 2020-10-22:
Is this crash is also occurring with camera 2 api?

Radcat501, 2020-10-22:
I dont know, but ill try and inform you about results.

Radcat501, 2020-10-22:
Still same result. I added app:cameraEngine="camera2" in my xml, and when i start video recording, i got this stacktrace again:

```
E/MediaRecorder: start failed: -19
E/MediaRecorder: stop called in an invalid state: 0
E/CameraEngine: onVideoResult result is null: something went wrong. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:314)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:425)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:605)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:201)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```

As result, video doesnt recorded.

achreffaidi, 2020-11-04:
@Radcat501 I am facing the same problem with Xiaomi Redmi note 7 running on Android 10 . Did you solve the problem ?

Radcat501, 2020-11-04:
@achreffaidi still not, but i suggest, the problem is in new File creating in Android versions above 8. Looks like File does not properly created, so CameraView cant start recording.

achreffaidi, 2020-11-04:
@Radcat501 in my case, it works for the Selfie camera but Fails for the Main Camera. for some version under 2.6.4 it fails for both.

achreffaidi, 2020-11-04:
And yesterday it randomly worked with version 2.3.0 and when i rebuilt the project today the problem is back

achreffaidi, 2020-11-06:
@Radcat501 Updates ! I found the reason behind that exception. In my case it's because i am running ExoPlayer in another activity and i guess both of them are using Hardware acceleration for video. my temporary fix was adding  
`
onPause(){
player.stop()
}`
to the activity that using ExoPlayer.
and that works for me ... but it created another problem when i try to run the media player again.

Radcat501, 2020-11-06:
@achreffaidi looks like in my case removing app:cameraVideoCodec="h263" removed the crash.

achreffaidi, 2020-11-06:
@PraveenOjha Following your steps, i fixed the problem by  setting  app:cameraVideoSizeBiggest="false" . I think recording the video using 48MP camera in Biggest video size is what required using the hardware acceleration . Though it's just a temporary fix.

achreffaidi, 2020-11-06:
> @achreffaidi looks like in my case removing app:cameraVideoCodec="h263" removed the crash.

@Radcat501  Following your steps, i fixed the problem by  setting  app:cameraVideoSizeBiggest="false" . I think recording the video using 48MP camera in Biggest video size is what required using the hardware acceleration . Though it's just a temporary fix.

closed 2020-12-12 by stale[bot]

coolxiao, 2022-08-03:
I also have this problem. Have you solved it @achreffaidi @Radcat501 thanks

krunalpatel3, 2023-05-15:
im facing this error when i upgrade android api level to 33.This error is coming in android 7 it is working well in android 13

E/MediaRecorder: start failed: -19
E/MediaRecorder: stop called in an invalid state: 0
E/CameraEngine: onVideoResult result is null: something went wrong. java.lang.RuntimeException: start failed.

  java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:314)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:426)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:605)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:84)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:81)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)