natario1/CameraView #979: Convert frame to ARGB8888

fanto88, 2020-10-22:
### How do I?
How do i convert a frame to ARGB8888?
I need to do an object detection using TensorFlow lite, im actually following this example [https://github.com/tensorflow/examples/tree/master/lite/examples/object_detection/android](url) but im not understanding how to achieve his conversion for the same model used in the example for every frame

This is the actual code:

` @Override
            @WorkerThread
            public void process(@NonNull Frame frame) {
                long time = frame.getTime();
                Size size = frame.getSize();
                int format = frame.getFormat();
                int userRotation = frame.getRotationToUser();
                int viewRotation = frame.getRotationToView();

                if (frame.getDataClass() == byte[].class) {
                    byte[] data = frame.getData();
                    // Process byte array...
                } else if (frame.getDataClass() == Image.class) {
                    Image data = frame.getData();
                    // Process android.media.Image...
                }
}`

PraveenOjha, 2020-10-23:
if (frame.getDataClass() == byte[].class) {
					byte[] data = frame.getData();
					Size s = frame.getSize();
					YuvImage yuv = new YuvImage(data, ImageFormat.NV21, s.getWidth(), s.getHeight(), null);
					ByteArrayOutputStream stream = new ByteArrayOutputStream();
					yuv.compressToJpeg(new Rect(0, 0, s.getWidth(), s.getHeight()), 100, stream);
					byte[] buf = stream.toByteArray();
					Int roatation=frame.getRotationToUser();
					Bitmap b =BitmapFactory.decodeByteArray(buf, 0, buf.length, null);
					// Process byte array...
				} else if (frame.getDataClass() == Image.class) {
					Image data = frame.getData();
					ByteBuffer buffer = data.getPlanes()[0].getBuffer();
					byte[] bytes = new byte[buffer.capacity()];
					buffer.get(bytes);
					Int r=frame.getRotationToUser();
					Bitmap b =BitmapFactory.decodeByteArray(bytes, 0, bytes.length, null);
				}
			}
		});

PraveenOjha, 2020-10-23:
See above code for bitmap form byre[] hope this help,you

fanto88, 2020-10-23:
Perfect Mate! Thank you for the help!

closed 2020-10-23 by fanto88