natario1/CameraView #986: Display problems on Android 11 (API 30)

JoAdams34, 2020-10-30:
On a Google Pixel 4a running Android 11 (API 30) the default cameraView (GL) display looks like this with a strange split into multiple panes:
[image]

polokoio456, 2020-11-04:
I test multiple Android 11 devices, only Pixel 4a has this problem.
I think it has nothing to do with Android 11.
Hoping someone for solving this issue.

polokoio456, 2020-11-04:
I test multiple Android 11 devices, only Pixel 4a has this problem. (Only back facing for me with pixel 4a)
Interesting, Pixel 4a's hardware supported level is LEVEL_FULL.

I found camera1 would not occur this issue.

Hoping someone for solving this issue.

polokoio456, 2020-11-05:
I found that I use camera2 and set the cameraExperimental to false, the camera work normally.

JoAdams34, 2020-11-06:
How can one use Camera2 without cameraExperimental? If you look at method "instantiateCameraEngine" in "CameraView.java" it says:
if (mExperimental && engine == Engine.CAMERA2 && Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) { return new Camera2Engine(callback); } else { mEngine = Engine.CAMERA1; return new Camera1Engine(callback); }

Otherwise it always uses Camera1, so the above implies that CameraView with Camera2 is broken on the Pixel4a.

closed 2020-12-12 by stale[bot]