natario1/CameraView #987: A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 24243 (FallbackCameraT)

android-dataticket, 2020-10-30:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.6.4
- Camera engine used: camera2
- Reproducible in official demo app: no
- Device / Android version: Samsung Galaxy S8+ / Android 9
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
The demo app switches to a preview of the photo after each snap, I don't think its viable to recreate this issue.
My app allows taking the photos rapidly, the photos are processed in the background and thumbnails are added to a GridView.
Also, this camera layout is *forced landscape*, so no matter what you tilt the phone, it's always in an official landscape layout.
I *think* this happens when holding the device in portrait, snapping photos (snapshots), and then holding it in landscape and snapping more photos (snapshots, always snapshots).  Again, no matter that the device is being held in portrait or landscape, the layout is in strict landscape mode, and the snapshots are cropped accordingly.

### Expected behavior
I expect the snapshots to continue to process, but randomly the SIGSEGV fatal signal gets thrown and the activity in which this camera resides is destroyed abruptly.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

     <com.otaliastudios.cameraview.CameraView
                android:id="@+id/camview"
                android:layout_width="match_parent"
                android:layout_height="match_parent"
                android:layout_gravity="center"
                android:adjustViewBounds="true"
                android:padding="5dp"
                app:cameraFacing="back"
                app:cameraFlash="off"
                app:cameraGesturePinch="zoom"
                app:cameraGestureTap="autoFocus"
                app:cameraMode="picture"
                app:cameraPictureFormat="jpeg" />

### Screenshots
no screenshot

### Logs (taking snapshots back to back, changing orientation for some)
1 CameraOrchestrator/FLASH (OFF) - Scheduling.
1 CameraOrchestrator/FLASH (OFF) - Executing.
1 CameraOrchestrator/FLASH (OFF) - Finished.
1 Angles/Angles changed: sensorOffset: 0 displayOffset: 90 deviceOrientation: 0
1 CameraEngine/START: scheduled. State: OFF
1 CameraOrchestrator/OFF >> ENGINE - Scheduling.
1 CameraOrchestrator/ENGINE >> BIND - Scheduling.
1 CameraOrchestrator/OFF >> ENGINE - Executing.
1 CameraOrchestrator/BIND >> PREVIEW - Scheduling.
1 CameraEngine/collectCameraInfo Facing: BACK Internal: 0 Cameras: 2
1 Angles/Angles changed: sensorOffset: 90 displayOffset: 90 deviceOrientation: 0
1 CameraCallbacks/onDeviceOrientationChanged 0
1 Angles/Angles changed: sensorOffset: 90 displayOffset: 90 deviceOrientation: 0
2 CameraView/doInstantiateEngine: instantiating. preview: GL_SURFACE
2 CameraView/doInstantiateEngine: instantiated. preview: GlCameraPreview
2 CameraView/onMeasure: surface is not ready. Calling default behavior.
2 CameraView/onMeasure: surface is not ready. Calling default behavior.
2 CameraView/onMeasure: surface is not ready. Calling default behavior.
2 CameraView/onMeasure: surface is not ready. Calling default behavior.
2 CameraView/onMeasure: surface is not ready. Calling default behavior.
2 CameraView/onMeasure: surface is not ready. Calling default behavior.
1 OverlayLayout/normal draw called.
1 CameraPreview/dispatchOnSurfaceAvailable: w= 1071 h= 634
1 CameraEngine/onSurfaceAvailable: Size is 1071x634
1 CameraOrchestrator/ENGINE >> BIND - Scheduling.
1 CameraOrchestrator/BIND >> PREVIEW - Scheduling.
1 CameraEngine/onStartEngine: Applying default parameters.
1 CameraEngine/onStartEngine: Ended
1 CameraOrchestrator/OFF >> ENGINE - Finished.
1 CameraOrchestrator/ENGINE >> BIND - Executing.
1 CameraCallbacks/dispatchOnCameraOpened com.otaliastudios.cameraview.engine.options.Camera1Options@f42a45b
[... 92 lines cut ...]
1 CameraEngine/takePictureSnapshot: running. isTakingPicture: false
1 CameraEngine/onTakePictureSnapshot: executing.
1 CameraEngine/onTakePictureSnapshot: executed.
1 CameraOrchestrator/TAKE PICTURE SNAPSHOT - Finished.
2 WorkerHandler/get: Reusing cached worker handler. FallbackCameraThread
1 SnapshotPictureRecorder/takeFrame: timestampUs: 6783156829
1 CameraCallbacks/dispatchOnPictureTaken com.otaliastudios.cameraview.PictureResult$Stub@98c5a84
1 CameraOrchestrator/TAKE PICTURE SNAPSHOT - Scheduling.
1 CameraOrchestrator/TAKE PICTURE SNAPSHOT - Executing.
1 CameraEngine/takePictureSnapshot: running. isTakingPicture: false
1 CameraEngine/onTakePictureSnapshot: executing.
1 CameraEngine/onTakePictureSnapshot: executed.
1 CameraOrchestrator/TAKE PICTURE SNAPSHOT - Finished.
2 WorkerHandler/get: Reusing cached worker handler. FallbackCameraThread
1 SnapshotPictureRecorder/takeFrame: timestampUs: 6783693994
1 CameraOrchestrator/TAKE PICTURE SNAPSHOT - Scheduling.
1 CameraOrchestrator/TAKE PICTURE SNAPSHOT - Executing.
1 CameraEngine/takePictureSnapshot: running. isTakingPicture: true
1 CameraOrchestrator/TAKE PICTURE SNAPSHOT - Finished.
*** CRASH HERE ***
A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 24243 (FallbackCameraT)
*** CRASH HERE ***

### APK
My apologies, our application is private :( I'll provide what I can upon request.

android-dataticket, 2020-11-02:
As probably already known, this error is not thrown when using the TextureView preview instead of OpenGL preview.

BubbleTrouble14, 2020-11-11:
I also seem to be getting this issue randomly. Doing live face recognition and drawing a rectangle around the face. I get the crash after navigating back from the fragment, but at random. Haven't been able to figure out why or when it occurs. Tried using TextureView, but that seemed to cause different issues.

closed 2020-12-12 by stale[bot]

devsontx, 2020-12-29:
Maybe my case is famillar with @BubbleTrouble14 's
It happened with both camera engine 1 and 2, CameraView 2.6.4, Picture mode.

It's observed on few models e.g Samsung J730G (Android 9) and Nokia 2.1 (Android 10) that working with tensorflow (any version) will cause crash after reporting `A/libc: Fatal signal 11 (SIGSEGV), code 1 ` 
No action like take snapshoot was needed to reproduce the error.

Susspected Tensorflow at first but then when I replace it by another CameraView library to produce frames, the error will disapear.

BubbleTrouble14, 2020-12-29:
I was actually able to fix it, was caused due to the face recognition library i was using. And also ended up not constantly drawing a rectangle around the face the whole time.

devsontx, 2020-12-29:
@BubbleTrouble14 may i ask what was the face recognition library. What is the root cause of your problem?

BubbleTrouble14, 2020-12-29:
Luxand Facerecogniton, yes I am about 80 percent sure. I was setting the size of an array of facetemplates incorrect which would cause memory leaks inside the library an cause a crash.

isabsent, 2021-03-21:
I catch this bug each time I trying to use Database Inspector in Android Studio under debug. It doesn't depend with CameraView usage. When I don't use Database Inspector this bug doesn't appear.

CalderonRigoberto, 2022-12-14:
@BubbleTrouble14 where you modify your settings? I'm strugglin with the same problem.

longpq-vmo, 2022-12-29:
Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xffffffff00000066 in tid 26770
me too, pls help me check issue when back fragment

longquangpham90, 2022-12-29:
Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xffffffff00000066 in tid 26770 
me too, pls help me check issue when back fragment

longpq-vmo, 2023-01-01:
Thanks all
I think issue by result bitmap screen 
Don't should use bitmap, we will save file data

longquangpham90, 2023-01-01:
Thanks all
I think issue by result bitmap screen
Don't should use bitmap, we will save file data