natario1/CameraView #988: takePictureSnapshot - Flash not working

abhilash007, 2020-10-31:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.6.4
- Camera engine used: both
- Device / Android version: Redmi 5 Pro

Flash not working on takePictureSnapshot()

android-dataticket, 2020-11-02:
It's a snapshot of the preview, you'd have to use the torch mode for flash

NaveenPokhriyal, 2020-11-25:
Auto Flash not working while clicking snapshot.

android-dataticket, 2020-11-25:
> Auto Flash not working while clicking snapshot.

you can't expect the snapshot to trigger flash.  snapshot is designed to be fast, quick, snappy.

if you want flash while using snapshot, you will need to use flash in torch mode, and toggle it manually on and off accordingly.

NaveenPokhriyal, 2020-11-25:
Is there any way to make autoflash run like while we set the focus for dim light surface?

android-dataticket, 2020-11-25:
> Is there any way to make autoflash run like while we set the focus for dim light surface?

You're defeating the purpose of the snapshot (speed) by making the flash toggle on and off automatically (it's much better, for the focus of the object being photographed, to have the flash toggled already with TORCH before using snapshot to take a photo), however if you were to do it, you could programmatically:

cameraView.setFlash(Flash.TORCH);
//maybe wait 200ms to ensure the flash has illuminated and the display preview has focused better on the illuminated object??
//Thread.sleep(200);
cameraView.takePictureSnapshot();
cameraView.setFlash(Flash.OFF);

NaveenPokhriyal, 2020-11-25:
Yes, I have achieved that flash functionality.
But can you help me with the auto flash issue. 
it didn't work when i set CaptureRequest.CONTROL_AE_MODE_ON_AUTO_FLASH for  -mOrignalaeMode.

NaveenPokhriyal, 2020-11-25:
I'm using the torch as Flash and later on setting that off and I also change action sequence for ( Actions.timeout(LOCK_TIMEOUT, new LockAction()), new FlashAction());) it handles flash on the exact surface.

closed 2020-12-31 by stale[bot]