natario1/CameraView #991: Null Surface targets are not allowed

mariuspena, 2020-11-09:
### Describe the bug
We get a lot of crashlytics reports like this one below
- CameraView version: *version number*
- Camera engine used: *camera1/camera2/both*
- Reproducible in official demo app: n/a
- Device / Android version: lots of devices, lots of android versions
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes/no*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to '...'
2. Click on '...'
3. See error

### Expected behavior
After updating the camera to camera2 engine these crashes started to appear.

### XML layout
we add it programatically like this

```
if (CameraUtils.hasCameras(this) == false) {
            throw new Exception("NO HARDWARE CAMERAS DETECTED!");
        }
        cameraView = new CameraView(this);
        cameraView.setAudio(Audio.OFF);
        cameraView.setHdr(Hdr.OFF);
        cameraView.setPlaySounds(false);
        cameraView.setMode(Mode.PICTURE);
        cameraView.setWhiteBalance(WhiteBalance.AUTO);
        cameraFacing = cameraView.getFacing();
        cameraView.addCameraListener(cameraListener);
        cameraView.setPreview(Preview.TEXTURE); // Circular reveal won't work otherwise
        cameraView.setExperimental(true);
        cameraView.setEngine(Engine.CAMERA2);
        if (BuildConfig.DEBUG) {
            CameraLogger.setLogLevel(CameraLogger.LEVEL_VERBOSE);
        }
        if (CameraUtils.hasCameraFacing(this, Facing.BACK))
            cameraView.setFacing(Facing.BACK);
        else if (CameraUtils.hasCameraFacing(this, Facing.FRONT))
            cameraView.setFacing(Facing.FRONT);
        else
            throw new Exception("NO FRONT OR BACK CAMERAS AVAILABLE!");
        // Camera result size is forced to 4:3 aspect ratio as it's the closest one to desired 5:4
        SizeSelector aspectRatio = SizeSelectors.aspectRatio(AspectRatio.of(4, 3), 0f);

        SizeSelector minWidth = SizeSelectors.minWidth(320);
        SizeSelector minHeight = SizeSelectors.minHeight(240);
        SizeSelector maxWidth = SizeSelectors.maxWidth(800);
        SizeSelector maxHeight = SizeSelectors.maxHeight(600);

        SizeSelector dimensionsMinLimitOnly = SizeSelectors.and(minWidth, minHeight);
        SizeSelector dimensions = SizeSelectors.and(minWidth, minHeight, maxWidth, maxHeight);
        SizeSelector min = SizeSelectors.smallest();
        SizeSelector requestSize = SizeSelectors.and(dimensions, min);
        SizeSelector requestSizeMinLimitOnly = SizeSelectors.and(dimensionsMinLimitOnly, min);

        SizeSelector result = SizeSelectors.or(
                SizeSelectors.and(aspectRatio, requestSize),             // Try to match all constraints
                requestSize,                                             // Try to match only size
                SizeSelectors.and(aspectRatio, requestSizeMinLimitOnly), // Try to match both aspect ratio and size with no upper limit
                requestSizeMinLimitOnly                                  // If none is found, at least try to match the request size with no upper limit
        );

        cameraView.setPictureSize(result);
        CameraOptions co = cameraView.getCameraOptions();
        if (co != null) {
            cameraView.setExposureCorrection(co.getExposureCorrectionMaxValue());
        }
        cameraFrame.addView(cameraView);
        cameraView.setLifecycleOwner(this);
```

### Logs
```
Fatal Exception: java.lang.IllegalArgumentException: Null Surface targets are not allowed
       at android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest(CameraDeviceImpl.java:1021)
       at android.hardware.camera2.impl.CameraDeviceImpl.setRepeatingRequest(CameraDeviceImpl.java:1094)
       at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:297)
       at com.otaliastudios.cameraview.engine.Camera2Engine.applyRepeatingRequestBuilder(Camera2Engine.java:249)
       at com.otaliastudios.cameraview.engine.Camera2Engine.onStartPreview(Camera2Engine.java:618)
       at com.otaliastudios.cameraview.engine.CameraEngine$10.call(CameraEngine.java:522)
       at com.otaliastudios.cameraview.engine.CameraEngine$10.call(CameraEngine.java:519)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(zzi.java:10)
       at com.google.android.gms.tasks.zzq.zza(zzq.java:19)
       at com.google.android.gms.tasks.zzu.zzb(zzu.java:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(zzi.java:10)
       at com.google.android.gms.tasks.zzq.zza(zzq.java:19)
       at com.google.android.gms.tasks.zzu.zza(zzu.java:104)
       at com.google.android.gms.tasks.zzd.onSuccess(zzd.java:8)
       at com.google.android.gms.tasks.zzn.run(zzn.java:4)
       at com.google.android.gms.tasks.zzt.execute(zzt.java:2)
       at com.google.android.gms.tasks.zzm.zza(zzm.java:11)
       at com.google.android.gms.tasks.zzq.zza(zzq.java:19)
       at com.google.android.gms.tasks.zzu.zzb(zzu.java:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigured(Camera2Engine.java:577)
       at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigured$0(CallbackProxies.java:53)
       at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$soW0qC12Osypoky6AfL3P2-TeDw.run(-.java:4)
       at android.os.Handler.handleCallback(Handler.java:907)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:216)
       at android.os.HandlerThread.run(HandlerThread.java:65)
```

mariuspena, 2020-11-26:
Can someone please help? We see dozens of this in crashlytics

Nir-Zigler, 2020-12-07:
Will appreciate some assistance with this issue.

Nir-Zigler, 2020-12-14:
Can we help with providing more details? what information is missing for further investigation?

Nir-Zigler, 2021-01-10:
Will appreciate any feedback on this issue. Is more information is required to investigate this issue? The majority (by far) of crashes occurs because of this issue.

closed 2021-06-26 by stale[bot]