natario1/CameraView #1006: java.lang.RuntimeException: start failed.

faldujaldeep, 2020-12-08:
### Describe the bug
I am getting the below exception while capturing video from the front camera.

- CameraView version: *2.6.4*
- Camera engine used: *camera1*
- Reproducible in official demo app: *yes*
- Device / Android version: *Android 5.0*

### To Reproduce
2020-12-08 20:29:28.529 22549-23548/? W/FullVideoRecorder: start: Error while starting media recorder. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:4)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:11)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:5)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:18)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:2)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:2)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:1)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:6)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:1)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)

 com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
     at com.otaliastudios.cameraview.engine.CameraBaseEngine.onVideoResult(CameraBaseEngine.java:4)
     at com.otaliastudios.cameraview.engine.Camera1Engine.onVideoResult(Camera1Engine.java:1)
     at com.otaliastudios.cameraview.video.VideoRecorder.dispatchResult(VideoRecorder.java:11)
     at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:20)
     at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:10)
     at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:9)
     at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:11)
     at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:5)
     at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:18)
     at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:2)
     at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:2)
     at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:1)
     at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:6)
     at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:1)
     at android.os.Handler.handleCallback(Handler.java:883)
     at android.os.Handler.dispatchMessage(Handler.java:100)
     at android.os.Looper.loop(Looper.java:214)
     at android.os.HandlerThread.run(HandlerThread.java:67)
 Caused by: java.lang.RuntimeException: start failed.
     at android.media.MediaRecorder.start(Native Method)
     at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:4)
 	... 12 more
E/FACE_RECOG: ---onCameraError---kotlin.Unit

### XML layout
This is xml configuration

```xml
<com.otaliastudios.cameraview.CameraView
            android:id="@+id/cameraView"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:layout_gravity="center"
            android:keepScreenOn="true"
            app:cameraAudio="off"
            app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
            app:cameraEngine="camera1"
            app:cameraExperimental="false"
            app:cameraFacing="front"
            app:cameraFlash="auto"
            app:cameraGestureLongTap="none"
            app:cameraGesturePinch="none"
            app:cameraGestureScrollHorizontal="none"
            app:cameraGestureScrollVertical="none"
            app:cameraGestureTap="none"
            app:cameraGrid="off"
            app:cameraHdr="off"
            app:cameraMode="video"
            app:cameraPlaySounds="false"
            app:cameraPreview="glSurface"
            app:cameraVideoMaxSize="3670016"
            app:cameraVideoSizeBiggest="false"
            app:cameraVideoSizeMinHeight="100"
            app:cameraVideoSizeMinWidth="100"
            app:cameraVideoSizeSmallest="true"
            app:cameraWhiteBalance="daylight" />
```

closed 2021-01-10 by stale[bot]

coolxiao, 2022-07-27:
I also encountered it. How did you solve it

coolxiao, 2022-08-05:
@faldujaldeep