natario1/CameraView #1008: FLASH NOT WORKING

blennerSilva, 2020-12-09:
### How do I?
How do i turn on the flash i tried evarything and it does not work.

### Version used
2.6.4

here is how im implementing it

 camera = findViewById(R.id.camera);
        camera.setLifecycleOwner(this);
        camera.setPictureMetering(true);
        camera.setHdr(Hdr.ON);
        camera.setEngine(Engine.CAMERA2);
        camera.setFlash(Flash.TORCH);
        camera.setWhiteBalance(WhiteBalance.AUTO);
        camera.setTouchscreenBlocksFocus(false);
        camera.mapGesture(Gesture.TAP, GestureAction.AUTO_FOCUS); // Tap to focus!
        camera.findFocus();

monikkacha, 2020-12-25:
i also tried but did not work , did you solved or still stuck ? i am trying to scan PDF417 with ML kit android and i need CameraView to turn on flash but did not work . i also tried to manually turn on flash but app is getting failed because camera is in use .

dkgupta2516, 2021-01-18:
Inside Fragment, camera flash not working

nauhalf, 2021-06-01:
I also tried but did not work.

Tested on Xiaomi Redmi Note 5

closed 2021-07-13 by stale[bot]