natario1/CameraView #1009: Camera2 not working on Pixel5

cpt23, 2020-12-10:
Camera2 API and GlCameraPreview do not work properly on Google Pixel 5 devices with latest OS (Android 11) and patch level. The display is split into multiple bands/blocks, regardless of settings (resolution, frame rate, shutter speed etc).
[image]

natario1, 2020-12-10:
Sorry to hear that but I don't have this device so there's nothing I can do.

natario1, 2021-01-03:
@cpt23 could you try with this PR https://github.com/natario1/CameraView/pull/1020 ?
It fixes a similar issue on Pixel 4.

To test, you have to modify the FpsRangeValidator class and add the Pixel 5 to the blacklist.

cpt23, 2021-01-06:
PR #1020 fixed the preview display issues on Google Pixel 5, thanks!

natario1, 2021-01-06:
So I should add "Google Pixel 5" to FpsRangeValidator right? Or did it fix the issue even without that change?

darrinps, 2021-02-19:
Think this can be closed? Just tested over lunch with the latest using a Pixel 5 and don't see this issue.