natario1/CameraView #1012: Egloo.kt line 60 com.otaliastudios.opengl.core.Egloo.checkEglError

xuanloct4, 2020-12-16:
### Describe the bug
Crash
- CameraView version: 2.6.4
- Camera engine used: both
- Reproducible in official demo app: no
- Device / Android version: Galaxy S8+ (Android 5.1.1)
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Open the app
2. Click on Open Camera button
3. Crash

### Expected behavior
App can open the CameraView

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<?xml version="1.0" encoding="utf-8"?>
<layout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto">

    <androidx.constraintlayout.widget.ConstraintLayout
        android:layout_width="match_parent"
        android:layout_height="match_parent">

        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/viewCamera"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toTopOf="parent"
            app:cameraUseDeviceOrientation="false"
            app:cameraPlaySounds="false"
            android:layout_width="match_parent"
            android:layout_height="match_parent"/>

        <com.ascend.money.universal.dr.feature.general.receiverinput.photo.CameraPreviewMask
            android:id="@+id/viewCameraMask"
            android:layout_width="match_parent"
            android:layout_height="0dp"
            android:background="#B3808080"
            app:layout_constraintBottom_toBottomOf="@+id/viewCamera"
            app:layout_constraintEnd_toEndOf="@+id/viewCamera"
            app:layout_constraintStart_toStartOf="@+id/viewCamera"
            app:layout_constraintTop_toTopOf="@+id/viewCamera" />

        <androidx.constraintlayout.widget.ConstraintLayout
            android:id="@+id/clHeader"
            android:layout_width="match_parent"
            android:layout_height="0dp"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintHeight_percent="0.2"
            app:layout_constraintStart_toStartOf="parent">

            <ImageView
                android:id="@+id/btnCapture"
                android:layout_width="@dimen/_42sdp"
                android:layout_height="@dimen/_38sdp"
                android:rotation="-90"
                app:srcCompat="@drawable/dr_ic_camera"
                android:background="?selectableItemBackgroundBorderless"
                app:layout_constraintBottom_toBottomOf="parent"
                app:layout_constraintEnd_toEndOf="parent"
                app:layout_constraintStart_toStartOf="parent"
                app:layout_constraintTop_toTopOf="parent" />

        </androidx.constraintlayout.widget.ConstraintLayout>

        <include
            android:id="@+id/loading"
            layout="@layout/base_view_screen_loading"
            android:visibility="gone" />
    </androidx.constraintlayout.widget.ConstraintLayout>

</layout>
```

### Screenshots
N.A

### Logs
Fatal Exception: java.lang.RuntimeException: Error during eglCreateContext (2): EGL error 0x3005
       at com.otaliastudios.opengl.core.Egloo.checkEglError(Egloo.kt:60)
       at com.otaliastudios.opengl.core.EglNativeCore.<init>(EglNativeCore.kt:78)
       at com.otaliastudios.opengl.core.EglCore.<init>(EglCore.kt:17)
       at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder.takeFrame(SnapshotGlPictureRecorder.java:184)
       at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder$2.run(SnapshotGlPictureRecorder.java:139)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.os.HandlerThread.run(HandlerThread.java:61)

### APK
N.A

Landi29, 2021-01-09:
I have used some time investigating the origin of this bug. As I see it, `at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder.takeFrame(SnapshotGlPictureRecorder.java:184)` points to the creation of an EglCore object in the `takeFrame()` method. `takeFrame()` is called on a worker thread on line 139 in `onRendererFrame` without exception handling, which causes the app to raise an exception and crash. What it is the approach in this case, how do we handle an exception from `takeFrame()`.

closed 2021-06-26 by stale[bot]

rohilla17, 2021-06-28:
I am also facing this issue. Please help

ankitgupta199, 2021-08-31:
@xuanloct4 - Have you got any solution for this issue, as I am using latest dependency, but still I am getting this crash.

kakarrot, 2022-03-21:
I am also facing this issue.

Crash logs below:

Fatal Exception: java.lang.RuntimeException: Error during eglCreateContext (2): EGL error 0x3006
       at com.otaliastudios.opengl.core.Egloo.checkEglError(Egloo.java:4)
       at com.otaliastudios.opengl.core.EglNativeCore.<init>(EglNativeCore.java:20)
       at com.otaliastudios.opengl.core.EglCore.<init>(EglCore.java:5)
       at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.e(TextureMediaEncoder.java:23)
       at com.otaliastudios.cameraview.video.encoding.MediaEncoder$1.run(MediaEncoder.java:7)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:224)
       at android.os.HandlerThread.run(HandlerThread.java:67)