natario1/CameraView #1014: API 30, preview images are rotated when moving from landscape activity to fragment in portrait activity

seungwooha-sw, 2020-12-22:
### Describe the bug
Rotate device to the landscape mode on a base activity -> start a new activity -> add a fragment containing CameraView -> set a portrait mode programmatically using 
```Kotlin 
requireActivity().requestedOrientation = ActivityInfo.SCREEN_ORIENTATION_PORTRAIT 
```
(device is still set as the landscape mode)
Then the CameraView shows 90 degree rotated previews

- CameraView version: 2.6.4
- Camera engine used: Camera1
- Reproducible in official demo app: no
- Device / Android version: Galaxy S20 , API 30

### Expected behavior
The CameraView shows previews without rotation 

### XML layout

```xml
<layout xmlns:android="http://schemas.android.com/apk/res/android">
      <androidx.constraintlayout.widget.ConstraintLayout
              android:layout_width="match_parent"
              android:layout_height="match_parent">

          <com.otaliastudios.cameraview.CameraView
                  android:id="@+id/camera_view"
                  android:layout_width="match_parent"
                  android:layout_height="match_parent" />
      </androidx.constraintlayout.widget.ConstraintLayout>
</layout>

```

### Screenshots
N/A

### Logs
N/A

### APK
N/A

rmi111, 2021-01-05:
Any update on this?

EuRatnikov, 2021-03-31:
Same issue. Can someone check, because it's really annoying.

closed 2021-07-13 by stale[bot]