natario1/CameraView #1018: Playing video as overlay to be captured

pierre-the-fighter-pilot, 2021-01-02:
Hi,  first, thank you for the excellent camera library.

I am developing an app to record sporting events and produce a video file that would look somewhat like a sports broadcast.

I have managed to get my scoreboard overlay working perfectly.  However, one of the features I am also trying to achieve is an instant replay.

I have 2 problems...  Evidently, I am trying to play a portion of the video that is currently being recorded, so that's going to be an issue...  I'll get there eventually.  The first issue I'm tying to fix now is to play a video (any video) as an overlay in CameraView and actually record it in the video snapshot.

I first tried using a VideoView with a test mp4 as overlay view.  I see the "replay" video played in the preview.  However, it's not being recorded in the video snapshot.  I also tried with a combination of TextureView and MediaPlayer and I got the same result.  Is there a solution to this (different view combo)?

Thanks,

Pierre

pierre-the-fighter-pilot, 2021-01-04:
I did some additional experiments with a SurfaceView this time.  Again, same result...  video visible in preview, but not recorded in the video snapshot.  So I tried an additional SurfaceView, with a semi-transparent background and this one shows up as expected.

I am suspecting that this may have to do with hardware acceleration.  I tried disabling it at the view level, but that didn't work.  Could just be that my call to [replay_video_view.setLayerType(View.LAYER_TYPE_SOFTWARE, null);] was ignored.  Of course this is only for testing as I suspect disabling hardware acceleration will be too slow to produce useful playback.

Also, Note that I have run my experiments on a virtual device so far.  Will also try on an actual device later.

Regards,

natario1, 2021-01-04:
I have never tried to do this, but I agree that you should not be able to draw anything hardware on the snapshot surface, because we use lockCanvas() internally. So no TextureView.

You can also try modifying the library to use lockHardwareCanvas() instead of lockCanvas() and at that point you might be able to use a TextureView

pierre-the-fighter-pilot, 2021-01-05:
Thanks for the advice.  I'll definitely give this a try, but it might take me some time to figure out how to import the library locally into my Java project so I can modify it.  I'm still very green with Android-Studio unfortunately...

Regards,

pierre-the-fighter-pilot, 2021-01-08:
I was finally able to import the cameraview module in my project.

lockHardwareCanvas() works.  Besides needing API 23 instead of 15, is there another drawback to this call?

Thanks,

natario1, 2021-01-08:
The hardware canvas does not have all draw capabilities, especially in older APIs. https://developer.android.com/guide/topics/graphics/hardware-accel 

In short, some Views might not render correctly. But we could add what you need in the library under a boolean flag, to make it opt-in for your specific use case.

pierre-the-fighter-pilot, 2021-01-09:
That sounds good.  I'm not sure on the internals here, but could it be view-based (for one specific overlay view) or set to hardware/software on the fly (i.e. as the video capture is being recorded)?

This solves the issue of playing the video as overlay.  However, I still have the issue of actually playing a past portion of the video snapshot as it's still being recorded.  It's probably not possible because of the encoding, but it might work if I had second video being recorded in parallel.  That video could be stopped and manipulated when a replay is requested.  It's probably beyond the scope of CameraView, but would it possible to a CameraView as an overlay within another CameraView?  Something like this:

<CameraView>
  <---Scoreboard overlay view---/>
  <CameraView>
    app:cameraFacing="back"
    <---Instant replay overlay view---/>
  </CameraView>
</CameraView>

The inner CameraView would make a series of video snapshots while the outer CameraView would not be tied up to a physical camera.  It would just be there to handle the overlays and make the master recording.

Thanks,

pierre-the-fighter-pilot, 2021-01-10:
I tried it...  long story short...  it didn't work.  I also tried some tweaks to get two video snapshots recorded at the same time and it didn't work either.  I think I would have to modify the stub to handle two files, but it wouldn't help me as I need to start/stop one of the recordings while the other one is ongoing.

I'll close the issue as the original query was resolved.

Thanks again,

closed 2021-01-10 by pierre-the-fighter-pilot