natario1/CameraView #1037: How do work correct  with rotation value of the photo?

nestoleh, 2021-01-22:
### How do I?
I can't understand how CameraView orient bitmap with a photo.

#### Example.
I need to take the HORIZONTAL photo and cut a rectangle with a specific ratio from it. Sample below

[image]

To do it I set activity landscape mode `requestedOrientation = ActivityInfo.SCREEN_ORIENTATION_LANDSCAPE` and disable orientation for CameraView `app:cameraUseDeviceOrientation="false"`

To get bitmap I use the next code:

```
  cameraView.addCameraListener(object : CameraListener() {
            override fun onPictureTaken(result: PictureResult) {
                cameraView.close()
                result.toBitmap { bitmap ->
                    if (bitmap == null) {
                        showPhotoSavingError()
                        cameraView.open()
                    } else {
                       // handle bitmap
                    }
                }
            }
        })
```

In most cases, the result bitmap is horizontal and looks exactly like on screen, but sometimes (is ~5% cases) it vertical, and my cut fragment is wrong (sample below)

[image]

The same happens if I need a vertical photo, in most cases all ok, but sometimes on some devices in SCREEN_ORIENTATION_PORTRAIT and with the disabled app:cameraUseDeviceOrientation saves the photo to bitmap as horizontal image rotated to -90 deg.

Of course, I can check the bitmap ratio and rotate it to the proper orientation, but I don't know if I need to rotate it clockwise or counterclockwise. How to save the photo in bitmap always in the same rotation as the user see it on the device?

### Version used
2.6.4
 
P/S/ Sorry for my bad English, I hope I explained the problem

jemshit, 2021-03-04:
Any answer? 
I get similar behavior on video taking with same config above. Mostly output is landscape, sometimes it is still portrait. 
What is desired is force the output to be landscape no matter how device is held. 
@nestoleh @natario1

nestoleh, 2021-03-04:
@jemshit I don't have a 100% correct solution, so currently I just rotate the picture to landscape (if height > width) before processing. Of course, there are can be some troubles because I don't know the correct angle for rotating (90 deg or -90 deg) so the result will be rotated to 180deg on some devices

closed 2021-06-26 by stale[bot]