natario1/CameraView #1043: Publish to Maven Central?

VGJohn, 2021-02-04:
[Starting from May 1, 2021 JCenter will stop working](https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/)

Describe the solution you'd like
CameraView artifacts should be available outside of JCenter, preferably in Maven Central.

Describe alternatives you've considered
[JitPack](https://jitpack.io/)
[GitHub Packages](https://docs.github.com/en/packages)

peterfortuin, 2021-03-07:
Is this going to happen?

reactivedroid, 2021-03-17:
@natario1 Since the day for shut down is coming near, is there an action plan for the migration? Happy to discuss

natario1, 2021-03-17:
JCenter will keep serving artifacts until February 2022, so we have 11 months. As soon as we have a new version rolling out and JCenter does not accept packages anymore, we'll move to maven central.

troy-lamerton, 2021-06-07:
@natario1 is there a rough ETA on the next release?

It's currently not possible to use jitpack.io for example, because of this error:

```
Execution failed for task ':golive:mergeReleaseResources'.
> Could not resolve all files for configuration ':golive:releaseRuntimeClasspath'.
   > Could not find com.otaliastudios.opengl:egloo:0.5.3.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/com/otaliastudios/opengl/egloo/0.5.3/egloo-0.5.3.pom
       - https://repo.maven.apache.org/maven2/com/otaliastudios/opengl/egloo/0.5.3/egloo-0.5.3.pom
       - http://jitpack.io/com/otaliastudios/opengl/egloo/0.5.3/egloo-0.5.3.pom
     Required by:
         project :golive > com.github.natario1:cameraview:v2.7.0
```

ergunkocak, 2021-06-10:
required library (egloo) moved to maven with version 0.6.0 :

https://github.com/natario1/Egloo/issues?q=maven

troy-lamerton, 2021-07-03:
resolved in https://github.com/natario1/CameraView/releases/tag/v2.7.1

this can be closed?

natario1, 2021-07-03:
Yes!

closed 2021-07-03 by natario1

VGJohn, 2021-07-03:
Thanks so much for getting this sorted 🥳

amar-w, 2021-09-20:
So from 2.7.1 it will be served via Maven Central? 

What about older versions if a developer, by any chance, is still using them?

natario1, 2021-09-20:
You can keep using jcenter for older versions until May 2022, if I remember correctly. After this, you will need to update.