natario1/CameraView #1045: The phone becomes stuck after denying permission

LG12300, 2021-02-06:
When I set'cameraRequestPermissions' to true, and denied the permission, the phone becomes very stuck.
[image]
### Device
Redmi k20
### Version used
2.7.0
### xml
```
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/cameraView"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        app:layout_constraintBottom_toTopOf="@id/v_bottom"
        app:layout_constraintEnd_toEndOf="parent"
        app:cameraPlaySounds="false"
        app:cameraAudio="off"
        app:cameraMode="picture"
        app:cameraRequestPermissions="true"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent" />
```

LG12300, 2021-02-06:
I have checked the problem, although this is not a bug, it is not a good idea to request permission in the 'onResume' function,because every time you request permission,'onPause' and'onResume' will be called, which will be an endless loop

closed 2021-06-21 by LG12300

xxxifan, 2021-08-05:
same issue.