natario1/CameraView #1061: NPE at CameraUtils.java line 93

muammil, 2021-03-09:
### Describe the bug
In my production app, I am getting this bug very often now(crashlytics), I have been using this library for almost 2 years but this issue was not there before, this one popped up very recently, in just about one month. Please look in to it

Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *1.7.0*
- Camera engine used: *both*
- Reproducible in official demo app: *not sure*
- Device / Android version: *Had this issue on xiaomi, oppo, samsung, vivo and other, android versions from 9 to 11*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

### Logs

> Fatal Exception: java.lang.NullPointerException: Attempt to get length of null array
>        at java.io.FilterOutputStream.write(FilterOutputStream.java:97)
>        at com.otaliastudios.cameraview.CameraUtils.writeToFile(CameraUtils.java:93)
>        at com.otaliastudios.cameraview.CameraUtils$1.run(CameraUtils.java:121)
>        at android.os.Handler.handleCallback(Handler.java:883)
>        at android.os.Handler.dispatchMessage(Handler.java:100)
>        at android.os.Looper.loop(Looper.java:237)
>        at android.os.HandlerThread.run(HandlerThread.java:67)

muammil, 2021-06-11:
I am still getting this issue in my production app, dev team please look in to it

closed 2021-07-21 by stale[bot]

MrChriZ, 2022-05-16:
Seeing the same here

javaherisaber, 2022-07-29:
@muammil @MrChriZ 
Has anyone managed to fix this issue?
I have the bug in production app

javaherisaber, 2022-07-29:
@natario1 
This issue is serious and causing crash on many devices
can you take a look and fix it?