natario1/CameraView #1064: Corrupted Video From Samsung J7 Neo

sachin-sivadasan, 2021-03-15:
### Corrupted Video
Video output is corrupted on samsung J7 Neo (SM-J701F)
- CameraView version: *2.7.0*
- Camera engine used: *both*
- Reproducible in official demo app: *yes*
- Device / Android version: *samsung J7 Neo (SM-J701F), API 28*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to main screen
2. Click on record 5sec video.
3. See the video is not playing well and duration of the video is more than 50min.

### Expected behavior
Video should play with duration of 5sec

### XML layout
Part of the XML layout from demo app (unchanged).
```xml
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="center"
        android:layout_marginBottom="88dp"
        android:keepScreenOn="true"
        app:cameraExperimental="true"
        app:cameraEngine="camera2"
        app:cameraPreview="glSurface"
        app:cameraPlaySounds="true"
        app:cameraGrid="off"
        app:cameraFlash="auto"
        app:cameraAudio="on"
        app:cameraFacing="back"
        app:cameraGestureTap="autoFocus"
        app:cameraGestureLongTap="none"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="filterControl1"
        app:cameraGestureScrollVertical="exposureCorrection"
        app:cameraMode="picture"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker">
    </com.otaliastudios.cameraview.CameraView>

```

### Screenshots

### Logs

### APK
Reproducible on demo app

sachin-sivadasan, 2021-03-26:
@natario1

HBiSoft, 2021-05-23:
I can't reproduce this on my Samsung J7 Neo

closed 2021-06-26 by stale[bot]