natario1/CameraView #1069: Add support for custom camera engines

ObsidianX, 2021-03-16:
### Problem to be addressed
Allow apps to supply their own CameraEngine implementations to support alternative camera sources such as USB cameras, videos, streaming sources, etc.

### Describe the solution you'd like
Add an API to provide a custom CameraEngine implementation.

### Additional context

natario1, 2021-03-16:
This has always been possible, just through a different mechanism - you subclass CameraView and override instantiateCameraEngine. You'll get the callbacks as well. Is there anything else that you'd need?

Direct API is easier, but it's like me and you using it, I prefer that an advanced use case like this has an advanced implementation path as well, since using CameraEngine requires familiarity with the source code.

ObsidianX, 2021-03-16:
Ah gotcha, that should work just fine. Thanks!

closed 2021-03-16 by ObsidianX