natario1/CameraView #1078: camera2 黑屏

RogueJun, 2021-04-12:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.6.2*
- Camera engine used: *camera2*
- Reproducible in official demo app: *yes*
- Device / Android version: *xiaomi 8, API 29*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. 设备小米8，MIUI 12.03版本中使用camera2进行拍照，在960x1280分辨率下预览显示为黑屏，报错值CameraDevice.StateCallback.ERROR_MAX_CAMERAS_IN_USE2

### Expected behavior
应该是正常预览并可以进行拍照

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
 <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:layout_gravity="center"
        android:layout_marginTop="@dimen/dp_48"
        android:keepScreenOn="true"
        app:cameraAudio="on"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
        app:cameraEngine="camera1"
        app:cameraExperimental="true"
        app:cameraFacing="back"
        app:cameraFlash="off"
        app:cameraGestureLongTap="none"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="exposureCorrection"
        app:cameraGestureScrollVertical="none"
        app:cameraGestureTap="autoFocus"
        app:cameraGrid="off"
        app:cameraMode="picture"
        app:cameraPlaySounds="true"
        app:cameraPreview="glSurface" />
```

cuibg, 2021-04-14:
this res's author is not a Chinese

cuibg, 2021-05-14:
[image]
哥们，你这个问题，就是没有连接成功，这个不是库的问题，你可以想办法处理下，不是什么复杂的问题，而且此库已经捕获了CameraException！
[image]
说明有其他的应用正在打开相机！
@RogueJun

closed 2021-07-13 by stale[bot]