natario1/CameraView #1080: Help making high quality video recording instant

szymonchaber, 2021-04-20:
We're using this library to record videos and the stability is great, but the delay between `takeVideo(File file)` and recording actually started is not short enough for us.

I made some changes which made it nearly instant, but the following issues emerged:
- Video recording from the front camera is rotated 180 degrees (upside-down);
- On some Xiaomi devices the screen is flickering green during recording;
- Overall smoothness is quite degraded

Could you look [at my changes to the library](https://github.com/szymonchaber/CameraView/pull/1) and help me identify which change causes what?
In our project, the CameraEngine2 class is copied verbatim and modified as seen. 
CameraView is extended and also modified as seen.

I would say my changes are pretty minimal, so I have no idea why the mentioned issues emerged.
Do you see any other problems with my modifications?

The link is not intended as a PR to this project (unless you can help me make it so).
If it inspires you to improve things then all the better!

Issues I read before asking this one:
#500
#414
#1036

mahmood199, 2021-04-21:
Can you help me mentioning what type of File should we pass in takeVideo function??
I went through the documentation but it did not mention this properly.
If you know what type of file we should pass in takeVideo() function then please help.

szymonchaber, 2021-04-21:
@mahmood199 I answered you in the issue you created #1081

szymonchaber, 2021-06-11:
@natario1 Definitely not stale :)

closed 2021-07-21 by stale[bot]

szymonchaber, 2022-02-03:
@natario1 Let's try again :) Can you look at this?