natario1/CameraView #1081: How do I record video?

mahmood199, 2021-04-21:
What should be the parameter in the takeVideo() function? and what does it mean
In the docs it say file
but I cant understand what type of file 
Can anyone give a example of how to record video with file parameter

szymonchaber, 2021-04-21:
Hi @mahmood199, the docs mean class `File`, so in Java:
```
File file = new File(context.filesDir, "filename.mp4");
cameraView.takeVideo(file);
```
Kotlin:
```
val file = File(context.filesDir, "filename.mp4")
cameraView.takeVideo(file)
```

Take a look [at this line in demo](https://github.com/natario1/CameraView/blob/bde88a946e1591dd23b066d59ffcc563c958e40a/demo/src/main/kotlin/com/otaliastudios/cameraview/demo/CameraActivity.kt#L277)

mahmood199, 2021-04-21:
Thank you for your response
Can I send the video stored in the result and send it to next activity where I will use another library to process it??

closed 2021-07-09 by mahmood199