natario1/CameraView #1088: Pixel 4a API 30 - camera view shows deformed image and has very slow framerate

honzasmuk, 2021-05-12:
### Describe the bug

- CameraView version: *v2.6.2 - v2.7.0*
- Camera engine used: *camera2*
- Reproducible in official demo app: *yes*
- Device / Android version: *Pixel 4a, API 30*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

In our application camera preview shows deformed image (multiple duplicate strips) and very low framerate. When tried the demo app on same device, the application crashes after few seconds.

Cameraview v2.6.1 works fine. The problem first appeared in v2.6.2 and is still present in later versions up to v2.7.0.

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Just run the demo app.
2. Wait.
3. See error.

### Screenshots
[image]

### Logs (from demo app with v2.7.0)
[cameraview-2.7.0-pixel4a.txt](https://github.com/natario1/CameraView/files/6465669/cameraview-2.7.0-pixel4a.txt)

natario1, 2021-05-12:
Thanks for reporting! Can you try adding your device to this list? https://github.com/natario1/CameraView/blob/master/cameraview/src/main/java/com/otaliastudios/cameraview/internal/FpsRangeValidator.java#L25

honzasmuk, 2021-05-12:
I tried to add `Google Pixel 4a` to `com.otaliastudios.cameraview.internal.FpsRangeValidator#sIssues` and the demo app works nice now - wow!

natario1, 2021-05-12:
Nice. Do you want to create a PR?

honzasmuk, 2021-05-12:
PR: https://github.com/natario1/CameraView/pull/1089

closed 2021-05-12 by natario1

honzasmuk, 2021-05-12:
thanks! when do you plan the next release?

natario1, 2021-05-12:
Probably in the weekend. Going to take a bit of work because we must migrate the library to Maven Central. Meanwhile you can avoid this issue by using the previewFrameRate option. You can try previewFrameRate=30, previewFrameRateExact=true