natario1/CameraView #1097: Could not resolve all files for configuration ':app:debugCompileClasspath'

Jayce-2177, 2021-06-09:
### Describe the bug
So I updated my android studio today and I am getting the lint warning "jcenter is at end of life", when I try to remove JCenter() as suggested by android studio I now get the following build failed message.

Execution failed for task ':app:dataBindingMergeDependencyArtifactsDebug'.
> Could not resolve all files for configuration ':app:debugCompileClasspath'.
   > Could not find com.otaliastudios:cameraview:2.7.0.
     Required by:
         project :app

Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.7.0
- Camera engine used: both
- Reproducible in official demo app: yes
- Device / Android version: N/A
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Use android studio latest version
2. Remove Jcenter from gradle
3. See error

### Expected behavior
Build to compile, I knew this was coming for awhile, but I have been waiting to see if this repo would be updated to accommodate this change.

joehy, 2021-06-12:
same problem

closed 2021-07-21 by stale[bot]

carlossilvaf, 2022-01-17:
same problem

HBiSoft, 2024-08-25:
@Jayce-2177 @joehy @carlossilvaf Did you resolve this issue? I'm currently facing the same issue.

HBiSoft, 2024-08-25:
Never mind, I fixed it by upgrading my Gradle build version using Upgrade Assistant. I also upgraded `api 'com.otaliastudios:cameraview:2.7.0'` to `api 'com.otaliastudios:cameraview:2.7.2'`