natario1/CameraView #1130: Crash Issue

vigneshwaranoptisol, 2021-09-01:
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.Camera2Engine.onStartBind(Camera2Engine.java:517)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.CameraEngine$8.call(CameraEngine.java:468)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.CameraEngine$8.call(CameraEngine.java:464)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34)
        at com.litpic.litpicsdkdroid.camerapreview.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:246)
        at android.os.HandlerThread.run(HandlerThread.java:67)

jvlppm, 2021-09-17:
This is happening fairly often here

```log
-16 23:12:35.137  6407 24446 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=1
09-16 23:12:35.140  6407 24446 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=1
09-16 23:12:35.140  6407 24446 I CameraService: disconnect: Disconnected client for camera 0 for PID 30166
09-16 23:12:35.144 30166 30166 E CameraEngine: EXCEPTION: Unexpected error! Throwing.
09-16 23:12:35.145 30166 30166 D AndroidRuntime: Shutting down VM
09-16 23:12:35.152 30166 30166 E AndroidRuntime: FATAL EXCEPTION: main
09-16 23:12:35.152 30166 30166 E AndroidRuntime: Process: com.mycameraapp.test, PID: 30166
09-16 23:12:35.152 30166 30166 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.Camera2Engine.onStartBind(Camera2Engine.java:517)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:470)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:466)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:938)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:223)
09-16 23:12:35.152 30166 30166 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:67)
```

```xml
<com.otaliastudios.cameraview.CameraView
                app:cameraMode="video"
                app:cameraVideoBitRate="6500000"
                android:id="@+id/camera"
                app:cameraFlash="off"
                app:cameraFacing="back"
                android:keepScreenOn="true"
                app:cameraEngine="camera2"
                app:cameraExperimental="true"
                android:layout_width="match_parent"
                android:layout_height="match_parent"/>
```

In my case I have two fragments one with CameraView on picture mode and other with video mode, and I swap between them

sometimes when entering the video fragment the image does not load and this error occurs

Android 11

Picture fragment reference:
```xml
<com.otaliastudios.cameraview.CameraView
                app:cameraMode="picture"
                android:id="@+id/camera"
                app:cameraFlash="off"
                app:cameraFacing="back"
                android:keepScreenOn="true"
                app:cameraEngine="camera2"
                app:cameraExperimental="true"
                android:layout_width="match_parent"
                android:layout_height="match_parent"/>
```

wisnuwiry, 2021-12-02:
How to solve this issue @natario1 ?

kakarrot, 2022-03-29:
I'm facing the same bug, How to solve this issue @natario1 ?

closed 2022-04-28 by stale[bot]

SweetD3v, 2022-10-13:
Tried some workarounds here & there, still getting so many crashes in the 'onStartBind()'.