natario1/CameraView #1141: preview and take picture is green screen in HUAWEI Harmony OS

gfzy9876, 2021-11-01:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.7.1
- Camera engine used: both
- Reproducible in official demo app: yes
- Device / Android version: HUAWEI(Harmony OS) - ELS-AN00(P40 Pro 5G), Api 29
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): 

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
```
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:layout_marginTop="103dp"
        android:layout_marginBottom="133dp"
        android:keepScreenOn="true"
        app:cameraExperimental="true"
        app:cameraEngine="camera2"
        app:cameraPreview="glSurface"
        app:cameraAudio="off"
        app:cameraFacing="front"
        app:cameraGestureTap="autoFocus" />
```

both using `app:cameraExperimental="true"`, `app:cameraEngine="camera2"`, `app:cameraPreview="glSurface"`

preview and take picture is this thing and with no error callback:

[image]

it just happen in Harmony OS

gfzy9876, 2021-11-15:
@natario1

kaitian521, 2022-01-10:
//        if (Build.VERSION.SDK_INT >= 21) {
//            sizeToProfileMap.put(new Size(3840, 2160),
//                    CamcorderProfile.QUALITY_2160P);
//        }

leocreater, 2022-01-11:
怎么处理的？我这也遇到了

gfzy9876, 2022-02-14:
@leocreater 
我把cameraEngine用camera1没问题了

```
<com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:layout_marginTop="103dp"
            android:layout_marginBottom="133dp"
            android:keepScreenOn="true"
            app:cameraEngine="camera1"
            app:cameraAudio="off"
            app:cameraFacing="front"
            app:cameraGestureTap="autoFocus" />
```

closed 2022-04-28 by stale[bot]

Qdafengzi, 2024-09-19:
anyone udpate?