natario1/CameraView #1142: Camera view crashes if mutiple child views are inlcluded in it

arsalanfakhar, 2021-11-01:
Camera view crashed when there are multiple child views are included inside of it 

For eg:

```
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:keepScreenOn="true"
        app:cameraEngine="camera2"
        app:cameraExperimental="true"
        app:cameraMode="video"
        app:cameraPreview="glSurface"
        tools:ignore="MissingClass">

        <com.views.OverlayView
            android:id="@+id/tracking_overlay"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:layout_alignParentStart="true"
            android:layout_alignParentTop="true"
            app:layout_drawOnPreview="true"
            app:layout_drawOnVideoSnapshot="true"
            android:layout_alignParentEnd="true" />

        <RelativeLayout
            android:id="@+id/containerParent"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            app:layout_drawOnPreview="true"
            app:layout_drawOnVideoSnapshot="true">

            <Button
                android:id="@+id/stopButton"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:layout_alignParentStart="true"
                android:layout_alignParentTop="true"
                android:layout_marginEnd="10dp"
                android:background="@color/colorBlack"
                android:text="Stop"
                android:textColor="@color/jogo"
                android:visibility="gone" />

            <Button
                android:id="@+id/restartButton"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:layout_alignParentTop="true"
                android:layout_toEndOf="@id/stopButton"
                android:background="@color/colorBlack"
                android:text="Restart"
                android:textColor="@color/jogo"
                android:visibility="gone" />

            <com.airbnb.lottie.LottieAnimationView
                android:id="@+id/gameHeading2"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:layout_gravity="center"
                android:gravity="center"
                app:lottie_autoPlay="true"
                app:lottie_fileName="game_heading.json"
                app:lottie_loop="true" />

            <Spinner
                android:id="@+id/gearSettings"
                android:layout_width="@dimen/dimen_50"
                android:layout_height="@dimen/dimen_50"
                android:layout_centerHorizontal="true"
                android:background="@drawable/gear"
                android:visibility="gone" />

        </RelativeLayout>

    </com.otaliastudios.cameraview.CameraView>
```

**Error occurred**
E: FATAL EXCEPTION: VideoEncoder
    Process: com.jogo, PID: 19988
    java.lang.IndexOutOfBoundsException: Index: 1, Size: 0
        at java.util.ArrayList.get(ArrayList.java:437)
        at android.view.ViewGroup.getAndVerifyPreorderedView(ViewGroup.java:4468)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4984)
        at android.view.View.draw(View.java:23057)
        at android.view.ViewGroup.drawChild(ViewGroup.java:5230)
        at com.otaliastudios.cameraview.overlay.OverlayLayout.doDrawChild(OverlayLayout.java:180)
        at com.otaliastudios.cameraview.overlay.OverlayLayout.drawChild(OverlayLayout.java:169)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4987)
        at com.otaliastudios.cameraview.overlay.OverlayLayout.drawOn(OverlayLayout.java:151)
        at com.otaliastudios.cameraview.overlay.OverlayDrawer.draw(OverlayDrawer.java:74)
        at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onFrame(TextureMediaEncoder.java:218)
        at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onEvent(TextureMediaEncoder.java:146)
        at com.otaliastudios.cameraview.video.encoding.MediaEncoder$3.run(MediaEncoder.java:244)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:237)
        at android.os.HandlerThread.run(HandlerThread.java:67)

Arsalanjogo, 2021-11-10:
@natario1

vaidik1011, 2022-04-01:
it crashes because you included animation view. i've also added image animation and it crashes and when i set it's visibility to "View.GONE" or "View.INVISIBLE" it works fine.  so basically it crashes when you add some sort of animation but another thing is it work completely fine on some devices like vivo v15, oppo and some other devices too..
@natario1  please check this out and tell us solution about it..

vaidik1011, 2022-04-28:
@natario1  please take a look on this issue