natario1/CameraView #1146: Image rotation on galaxy A32 and other samsung devices

leandro1995, 2021-11-14:
The image rotates when taking the photo and saving it but with other equipment this case does not happen this case could be an internal error of the library, the library is the last 2.7.2 is the one that happened in the case

undmeer, 2022-01-06:
May be a device bug.
Please refer to the following link,
[android - Photo rotated from camera (SAMSUNG device) - Stack Overflow](https://stackoverflow.com/questions/47261434/photo-rotated-from-camera-samsung-device)

to fix:
```
camera.addCameraListener(new CameraListener() {
    @Override
    public void onPictureTaken(PictureResult result) {
        if (result.getRotation() != 0) {
            //Rotation occurred.
            //If it wasn't intended,
            //Rerotate the image.
        }
    }
});
```

leandro1995, 2022-01-07:
@undmeer I verified it and it is not a device error there is a utility called CameraUtils.decodeBitmap from the same library that leaves it normal, the problem I think a validation is internal to the library or it is not being considered

closed 2022-04-28 by stale[bot]