natario1/CameraView #1147: Callback for preview stream availability

razlesh1, 2021-11-15:
I am looking for a way to observe preview stream availability, meaning when is the preview shown, and when it is closed.
Reading the documentation of CameraView, I haven't find a way to do so. Does this feature exist in CameraView? or this behavior can be implemented somehow? and if so then how?

### Version Used
CameraView 2.7.2

natario1, 2021-11-16:
There's no API for this unfortunately.

razlesh1, 2021-11-24:
Will there be in a future CameraView release?
I know that CameraX has that option and it is using Camera2 so I believe this behavior can be achieved somehow.
Anyways, thanks for the quick answer!

natario1, 2021-11-24:
There might be, it's not hard to do. If you plan to work on it, I can point you to the right place.

razlesh1, 2021-11-25:
It'd be great if you could point me to the right place🙏

natario1, 2021-11-25:
I would do this:

1. create a new sealed class for the camera state, something like:

   ```
   sealed class CameraState { 
      class Closed : CameraState()
      class Open : CameraState()
      class Preview : CameraState()
   }
   ```
   Let's use a sealed class instead of enum because in the future we will add more information.
  
3. add `onCameraStateChanged(@NonNull CameraState state)` to `CameraListener`
4. in [CameraEngine.Callback](https://github.com/natario1/CameraView/blob/main/cameraview/src/main/java/com/otaliastudios/cameraview/engine/CameraEngine.java#L116-L117), add a new method like dispatchCameraState
5. implement this method in CameraView, you can use the other dispatch* methods as reference https://github.com/natario1/CameraView/blob/main/cameraview/src/main/java/com/otaliastudios/cameraview/CameraView.java#L2224
6. we already have another CameraState, `com.otaliastudios.cameraview.engine.orchestrator.CameraState`. Please rename this to `CameraInternalState` to avoid confusion.
7. We need to be able to convert internal state to the new CameraState. It's easy, OFF means Closed(), ENGINE and BIND mean Open(), PREVIEW means Preview().
8. Add a new function [here](https://github.com/natario1/CameraView/blob/main/cameraview/src/main/java/com/otaliastudios/cameraview/engine/orchestrator/CameraOrchestrator.java#L39-L43), like `onInternalStateChanged(CameraInternalState)`. The function should be invoked when the internal state changes, here: https://github.com/natario1/CameraView/blob/main/cameraview/src/main/java/com/otaliastudios/cameraview/engine/orchestrator/CameraStateOrchestrator.java#L74
9. Last step: whenever internal state changes, dispatch the new state. This should be done [here](https://github.com/natario1/CameraView/blob/main/cameraview/src/main/java/com/otaliastudios/cameraview/engine/CameraEngine.java#L143-L152), implement the new method that you added in 8., convert internal state to camera state as we described, and pass to callback defined in 4. `mCallback.dispatchCameraState(internalState.toCameraState())` 

Then run the app and verify that it works :-)