natario1/CameraView #1157: SIGABRT encountered on writing an invalid file descriptor

lancewoo, 2021-12-25:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.7.2*
- Camera engine used: *camera2*
- Reproducible in official demo app: *no*
- Device / Android version: *Pie, API 25*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

The scenario is like this. We had a timer set for taking pictures every 5 minutes. Occasionaly the app would get a SIGABRT to be killed when creating a new CameraView fragment for taking a new picture. After checking the logcat and stacktrace, we suspect that there might be something wrong with the states of CameraOrchestrator or WorkerHandler. So we passed true to recreateHandler in the CameraEngine constructor and **never** got crashed again.

https://github.com/natario1/CameraView/blob/00579814084fdc793fcf7246a00e5b05286ccd1f/cameraview/src/main/java/com/otaliastudios/cameraview/engine/CameraEngine.java#L158

My question is, is it possible that WorkerHandler used an old thread while its Looper had quit?

### Expected behavior
A clear and concise description of what you expected to happen.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
  <com.otaliastudios.cameraview.CameraView xmlns:app="http://schemas.android.com/apk/res-auto"
        android:id="@+id/camera_view"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_centerInParent="true"

        android:keepScreenOn="true"
        app:cameraAudio="off"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
        app:cameraEngine="camera2"
        app:cameraFlash="auto"
        app:cameraGestureLongTap="takePicture"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="exposureCorrection"
        app:cameraGestureScrollVertical="none"
        app:cameraGestureTap="autoFocus"
        app:cameraGrid="off"
        app:cameraWhiteBalance="auto"
        app:cameraPlaySounds="false"
        app:cameraPreview="glSurface">

    </com.otaliastudios.cameraview.CameraView>
```

### Logs
Use `CameraLogger.setLogLevel(LEVEL_INFO)` to see all logs into LogCat.

12-23 11:09:34.589  4590  4590 I #ST2303_CameraServer#: [(Camera2Fragment.java:256)#onCreateView] Camera2Fragment onCreateView(id=1)
12-23 11:09:34.602  4590  4590 W CameraView: doInstantiateEngine: instantiating. engine: CAMERA2
12-23 11:09:34.602  4590  4590 W WorkerHandler: get: Reusing cached worker handler. CameraViewEngine
12-23 11:09:34.603  4590  4590 W CameraView: doInstantiateEngine: instantiated. engine: Camera1Engine
12-23 11:09:34.603  4590  4590 I CameraOrchestrator: FACING - Scheduling.
12-23 11:09:34.604  4590  4590 F Looper  : Could not write wake signal to fd 56: Bad file descriptor

pid: 4298, tid: 4298, name: er.cameraserver  >>> com.senter.cameraserver <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: 'Could not write wake signal to fd 56: Bad file descriptor'
    r0  00000000  r1  000010ca  r2  00000006  r3  00000008
    r4  000010ca  r5  000010ca  r6  ffb304ac  r7  0000010c
    r8  00000000  r9  efe48000  r10 12dc0f38  r11 00000000
    ip  f23a33cc  sp  ffb30498  lr  f230ef95  pc  f2305e5e

backtrace:
    #00 pc 0001ce5e  /system/lib/libc.so (abort+58)
    #01 pc 00007e19  /system/lib/liblog.so (__android_log_assert+156)
    #02 pc 0000f8d3  /system/lib/libutils.so (android::Looper::wake()+106)
    #03 pc 003abc5f  /system/framework/arm/boot-framework.oat (offset 0x3ab000) (android.app.backup.BackupDataInput.dtor [DEDUPED]+102)
    #04 pc 000123e7  <anonymous:d6d2e000> (android.os.MessageQueue.enqueueMessage+462)
    #05 pc 0001e7a5  <anonymous:d6d2e000> (android.os.Handler.enqueueMessage+100)
[... 225 lines cut ...]
    #231 pc 003e18cb  /system/lib/libart.so (MterpInvokeVirtual+442)
    #232 pc 003ffe94  /system/lib/libart.so (ExecuteMterpImpl+14228)
    #233 pc 0038f0a6  /system/framework/boot-framework.vdex (android.app.ActivityThread.handleResumeActivity+12)
    #234 pc 001c4d73  /system/lib/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.850926792+378)
    #235 pc 001c9459  /system/lib/libart.so (art::interpreter::ArtInterpreterToInterpreterBridge(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame*, art::JValue*)+152)
    #236 pc 001e05df  /system/lib/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+790)
    #237 pc 003e18cb  /system/lib/libart.so (MterpInvokeVirtual+442)
    #238 pc 003ffe94  /system/lib/libart.so (ExecuteMterpImpl+14228)
    #239 pc 003d1044  /system/framework/boot-framework.vdex (android.app.servertransaction.ResumeActivityItem.execute+24)
    #240 pc 001c4d73  /system/lib/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.850926792+378)
    #241 pc 001c9459  /system/lib/libart.so (art::interpreter::ArtInterpreterToInterpreterBridge(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame*, art::JValue*)+152)
    #242 pc 001e05df  /system/lib/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+790)
    #243 pc 003e18cb  /system/lib/libart.so (MterpInvokeVirtual+442)
    #244 pc 003ffe94  /system/lib/libart.so (ExecuteMterpImpl+14228)
    #245 pc 003d1640  /system/framework/boot-framework.vdex (android.app.servertransaction.TransactionExecutor.executeLifecycleState+104)
    #246 pc 001c4d73  /system/lib/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.850926792+378)
    #247 pc 001c9459  /system/lib/libart.so (art::interpreter::ArtInterpreterToInterpreterBridge(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame*, art::JValue*)+152)
    #248 pc 001e05df  /system/lib/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+790)
    #249 pc 003e2759  /system/lib/libart.so (MterpInvokeDirect+196)
    #250 pc 003fff94  /system/lib/libart.so (ExecuteMterpImpl+14484)
    #251 pc 003d14a4  /system/framework/boot-framework.vdex (android.app.servertransaction.TransactionExecutor.execute+74)
    #252 pc 001c4d73  /system/lib/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.850926792+378)
    #253 pc 001c9459  /system/lib/libart.so (art::interpreter::ArtInterpreterToInterpreterBridge(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame*, art::JValue*)+152)
    #254 pc 001e05df  /system/lib/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+790)
    #255 pc 003e18cb  /system/lib/libart.so (MterpInvokeVirtual+442)

lancewoo, 2021-12-29:
This seems to be a system level bug in Looper. This crash happened when a WorkerHandler thread was created and cached and a new CameraView instance was created and it was awaken up by an alarm.
I finally changed how CameraEngine use the WorkerHandler to do a workaround. Whenever recreateHandler() is called, a new key is used to create a new WorkerHandler; when CameraEngine quits, the WorkerHandler quits then.

natario1, 2021-12-29:
Thanks for the information. I don't fully understand what's going on though. Is there something the library could do, other than creating a new handler each time.

lancewoo, 2022-01-04:
I strongly suspect there's something wrong with my version of Android Looper at the very core. [https://android.googlesource.com/platform/system/core/+log/master/libutils/Looper.cpp](url)

More information, the device we're using is for surveillance purpose on power towers. There's no user interaction. So we only set up alarms to do picture shots repeatedly, say, every 30 minutes. After each picture taken, the device would go into sleep mode to save battery as soon as possible. A crash may occur on each second wakeup to do a picture taking. If we delay some time, say, 100ms or more, to do the instantiation of CameraView, it seems the crash case would not be touched. Weird? Since this is probably a bug of the underlying Android system, as we can see from stacktraces of each crash, I don't think there is a need to do modifications to the CameraView. It is a great library. Thanks! Moreover, this crash case only occurred to an Android 9 device, not on other kinds.