natario1/CameraView #1158: Can the camera be prepared in other thread to avoid UI glitches while opening?

usmanrana-ilsa, 2022-01-25:
### Version used
2.7.2

anggit97, 2022-04-21:
@usmanrana-ilsa do you have solve this problem?

usmanrana07, 2022-04-21:
In my case I was doing some other operations too during swipe on page. Holding them unless camera fragment is resumed improved the performance. Otherwise camera is preparing in main thread and that would cause some glitch. Almost similar behavior is observed in Instagram camera as well.

anggit97, 2022-04-21:
im using view pager with 3 fragments. when page is swiped, i'm not doing any  operation. but its still show some glitch (showing overlay gray, before camera preview is loaded). do you experience it?

usmanrana07, 2022-04-21:
yes, CameraView isn't resuming from last paused preview, but shows shuttering effect while resume again

tashaerxing, 2022-04-24:
How to solve the problem of slow focus when opening？