natario1/CameraView #1166: TakeVideo returns a corrupted file (green video) with a Huawei Android 10

MkGrimaldos, 2022-03-31:
### Describe the bug
When you take a video using Camera2 and with `cameraVideoSizeAspectRatio="1:1"` in the CameraView, it returns a corrupted file (green video) with a Huawei Android 10

- CameraView version: v2.7.2
- Camera engine used: camera2
- Reproducible in official demo app: yes
- Device / Android version: Huawei Honor 8X (JSN-L21), EMUI 10.0 API 29
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. With `cameraVideoSizeAspectRatio="1:1"` and `app:cameraEngine="camera2"` in the `activity_camera.xml`
2. Click on Edit and select Mode Video 
3. Click on captureVideo
4. See error (video preview is a corrupted green video)

### Expected behavior
Video preview should show a normal video

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<com.otaliastudios.cameraview.CameraView
    android:id="@+id/camera"
    android:layout_width="wrap_content"
    android:layout_height="wrap_content"
    android:layout_gravity="center"
    android:layout_marginBottom="88dp"
    android:keepScreenOn="true"
    app:cameraExperimental="true"
    app:cameraEngine="camera2"
    app:cameraPreview="glSurface"
    app:cameraPlaySounds="true"
    app:cameraGrid="off"
    app:cameraFlash="auto"
    app:cameraAudio="on"
    app:cameraFacing="back"
    app:cameraGestureTap="autoFocus"
    app:cameraGestureLongTap="none"
    app:cameraGesturePinch="zoom"
    app:cameraGestureScrollHorizontal="filterControl1"
    app:cameraGestureScrollVertical="exposureCorrection"
    app:cameraMode="picture"
    app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
    app:cameraVideoSizeAspectRatio="1:1">
...
</CameraView>
```

### Screenshots
[image]

1136346879, 2023-08-16:
hello， Is the issue solved? I now have a green screen when opening the preview on my Huawei phone

1136346879, 2023-08-16:
[image]

1136346879, 2023-08-16:
[image]