natario1/CameraView #1167: CameraEngine.java line 313 com.otaliastudios.cameraview.engine.CameraEngine.destroy

chnovo29, 2022-04-25:
### 
- CameraView version: *2.7.2*
- Camera engine used: *both*
- Device / Android version: *realme 7 Pro*
- Bug Type: *ANR*

### XML layout

```xml
 <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:keepScreenOn="true"
            app:cameraAudio="off"
            app:cameraFlash="auto"
            app:cameraGesturePinch="zoom"
            app:cameraGestureTap="autoFocus"
            app:cameraMode="picture"
            app:cameraSnapshotMaxHeight="1000"
            app:cameraSnapshotMaxWidth="1000"
           />
```

### Logs
main (timed waiting): tid=1 systid=25507 
       at sun.misc.Unsafe.park(Native method)
       at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:230)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer.doAcquireSharedNanos(AbstractQueuedSynchronizer.java:1090)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer.tryAcquireSharedNanos(AbstractQueuedSynchronizer.java:1390)
       at java.util.concurrent.CountDownLatch.await(CountDownLatch.java:278)
       at com.otaliastudios.cameraview.engine.CameraEngine.destroy(CameraEngine.java:313)
       at com.otaliastudios.cameraview.engine.CameraEngine.destroy(CameraEngine.java:325)
       at com.otaliastudios.cameraview.engine.CameraEngine.destroy(CameraEngine.java:289)
       at com.otaliastudios.cameraview.CameraView.destroy(CameraView.java:886)
       at java.lang.reflect.Method.invoke(Native method)
       at androidx.lifecycle.ClassesInfoCache$MethodReference.invokeCallback(ClassesInfoCache.java:222)
       at androidx.lifecycle.ClassesInfoCache$CallbackInfo.invokeMethodsForEvent(ClassesInfoCache.java:199)
       at androidx.lifecycle.ClassesInfoCache$CallbackInfo.invokeCallbacks(ClassesInfoCache.java:190)
       at androidx.lifecycle.ReflectiveGenericLifecycleObserver.onStateChanged(ReflectiveGenericLifecycleObserver.java:40)
       at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent(LifecycleRegistry.java:354)
       at androidx.lifecycle.LifecycleRegistry.backwardPass(LifecycleRegistry.java:284)
       at androidx.lifecycle.LifecycleRegistry.sync(LifecycleRegistry.java:302)
       at androidx.lifecycle.LifecycleRegistry.moveToState(LifecycleRegistry.java:148)
       at androidx.lifecycle.LifecycleRegistry.handleLifecycleEvent(LifecycleRegistry.java:134)
       at androidx.lifecycle.ReportFragment.dispatch(ReportFragment.java:68)
       at androidx.lifecycle.ReportFragment$LifecycleCallbacks.onActivityPreDestroyed(ReportFragment.java:224)
       at android.app.Activity.dispatchActivityPreDestroyed(Activity.java:1551)
       at android.app.Activity.performDestroy(Activity.java:8395)
       at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1345)
       at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:5439)
       at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:5483)
       at android.app.servertransaction.DestroyActivityItem.execute(DestroyActivityItem.java:44)
       at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:187)
       at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:102)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2303)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:254)
       at android.app.ActivityThread.main(ActivityThread.java:8231)
       at java.lang.reflect.Method.invoke(Native method)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:612)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1006)

imagineDev, 2022-06-10:
Facing the same issue, mostly on Mi/ Redmi devices.
Any fix for this?

Fighter2b, 2022-11-23:
Same issue, Any idea how to fix this ?

WangcWj, 2023-01-06:
Same issue, Any idea how to fix this ?

qinRight, 2023-05-17:
If you used 'cameraView.setLifecycleOwner(this)', do not call 'cameraview.destory()', otherwise 'cameraview.destory() ' will be called repeatedly. You can try it out.

ShakeelAziz, 2023-07-26:
I've used cameraView.setLifecycleOwner(viewLifecycleOwner) in fragment and no destroy code is written in onDestroy() or onResume() but still gives me ANR of CameraEngine.destroy when I move to next fragment from camera fragment and back to Camera. How to Fix this?

yikeng, 2023-11-14:
In your fragment : cameraView.setLifecycleOwner(viewLifecycleOwner) 
do not use: cameraView.setLifecycleOwner(this)

BruceLiuTao, 2023-12-08:
Same issue, Any idea how to fix this ?

gi13371, 2024-01-09:
Same issue，have no idea to fix it!@natario1

EzequielAdrianM, 2024-09-16:
Yeah, same here. Looks like CameraEngine's got a CountDownLatch of 6 seconds that somehow is (accidentally?) running on the UI thread, thus causing ANR. @natario1 

```  at jdk.internal.misc.Unsafe.park (Native method)
  at java.util.concurrent.locks.LockSupport.parkNanos (LockSupport.java:252)
  at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquire (AbstractQueuedSynchronizer.java:717)
  at java.util.concurrent.locks.AbstractQueuedSynchronizer.tryAcquireSharedNanos (AbstractQueuedSynchronizer.java:1074)
  at java.util.concurrent.CountDownLatch.await (CountDownLatch.java:276)
  at com.otaliastudios.cameraview.engine.CameraEngine.destroy (CameraEngine.java:313)
  at com.otaliastudios.cameraview.engine.CameraEngine.destroy (CameraEngine.java:325)
  at com.otaliastudios.cameraview.engine.CameraEngine.destroy (CameraEngine.java:289)
  at com.otaliastudios.cameraview.engine.CameraEngine$2.run (CameraEngine.java:235)
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loopOnce (Looper.java:217)
  at android.os.Looper.loop (Looper.java:309)
  at android.app.ActivityThread.main (ActivityThread.java:8093)
  at java.lang.reflect.Method.invoke (Native method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:579)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1045)```

natario1, 2024-09-16:
it's not accidental I fear, I remember that if `destroy()` doesn't ensure that everything is cleaned up synchronously, many bugs appear during navigation or other scenarios like config changes, where the old engine is still running and the new engine fails to connect because of this.

One could implement better solutions of course.

khiemdnt, 2024-12-30:
> Yeah, same here. Looks like CameraEngine's got a CountDownLatch of 6 seconds that somehow is (accidentally?) running on the UI thread, thus causing ANR. @natario1
> 
> ```
>   at java.util.concurrent.locks.LockSupport.parkNanos (LockSupport.java:252)
>   at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquire (AbstractQueuedSynchronizer.java:717)
>   at java.util.concurrent.locks.AbstractQueuedSynchronizer.tryAcquireSharedNanos (AbstractQueuedSynchronizer.java:1074)
>   at java.util.concurrent.CountDownLatch.await (CountDownLatch.java:276)
>   at com.otaliastudios.cameraview.engine.CameraEngine.destroy (CameraEngine.java:313)
>   at com.otaliastudios.cameraview.engine.CameraEngine.destroy (CameraEngine.java:325)
>   at com.otaliastudios.cameraview.engine.CameraEngine.destroy (CameraEngine.java:289)
>   at com.otaliastudios.cameraview.engine.CameraEngine$2.run (CameraEngine.java:235)
>   at android.os.Handler.handleCallback (Handler.java:938)
>   at android.os.Handler.dispatchMessage (Handler.java:99)
>   at android.os.Looper.loopOnce (Looper.java:217)
>   at android.os.Looper.loop (Looper.java:309)
>   at android.app.ActivityThread.main (ActivityThread.java:8093)
>   at java.lang.reflect.Method.invoke (Native method)
>   at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:579)
>   at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1045)```
> ```

Same issue. @EzequielAdrianM Do you have a solution to fix this issue?

lterminiello, 2025-02-18:
Any solution for this issue? We have the same ANR in customview onDestory call
```
    @OnLifecycleEvent(Lifecycle.Event.ON_DESTROY)
    fun onDestroy() {
        camera?.let {
            it.clearCameraListeners()
            it.destroy()
            camera = null
        }
    }
```