natario1/CameraView #1171: NullpointerException on onStartBind()

SweetD3v, 2022-05-24:
###
Hi tech folks. I am getting this bug again and again in some devices. I tried many times to solve that, but it's still happening for some reason. Anyone having this issue? then please help me solve it.

- CameraView version: 2.7.2
- Camera engine used: camera2
- Reproducible in official demo app: no
- Device / Android version: Samsung devices mostly with API 30+
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): yes

### To Reproduce
It's not reproduceable sadly, I am getting it from the play store app although I didn't apply any proguard.

### XML layout
XML Layout :

```xml
<com.otaliastudios.cameraview.CameraView
                android:id="@+id/cameraView"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:layout_gravity="center"
                android:animateLayoutChanges="true"
                android:keepScreenOn="true"
                app:cameraAudio="on"
                app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
                app:cameraEngine="camera2"
                app:cameraExperimental="true"
                app:cameraFacing="back"
                app:cameraGestureLongTap="none"
                app:cameraGesturePinch="zoom"
                app:cameraGestureScrollVertical="exposureCorrection"
                app:cameraHdr="on"
                app:cameraMode="picture"
                app:cameraPictureMetering="false"
                app:cameraPlaySounds="false"
                app:cameraPreview="glSurface"
                app:cameraPreviewFrameRate="30"
                app:cameraPreviewFrameRateExact="true"
                app:cameraRequestPermissions="false"/>
```

### Logs
`java.lang.NullPointerException: 
  at com.otaliastudios.cameraview.engine.Camera2Engine.onStartBind (Camera2Engine.java)
  at com.otaliastudios.cameraview.engine.CameraEngine$2.call (CameraEngine.java:22)
  at com.otaliastudios.cameraview.engine.CameraEngine$2.call (CameraEngine.java)
  at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$1.call (CameraStateOrchestrator.java)
  at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$1.call (CameraStateOrchestrator.java)
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run (CameraOrchestrator.java:4)
  at com.otaliastudios.cameraview.internal.WorkerHandler.run (WorkerHandler.java:10)
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute (CameraOrchestrator.java:13)
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.lambda$sync$0 (CameraOrchestrator.java:46)
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.$r8$lambda$HmE6GLzVomrx4z2yFPqPM3-OldU (CameraOrchestrator.java)
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$$InternalSyntheticLambda$0$8b677079dbbaae903a74c190d1ab73dae346936de4769c0e3a915f12c512d38e$0.run (CameraOrchestrator.java:2)
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loop (Looper.java:236)
  at android.os.HandlerThread.run (HandlerThread.java:67)`

razlesh1, 2022-06-14:
Having a similar issue issue.
Anyone solved it?

razlesh1, 2022-06-15:
Hi, thank you for answering.
I didn't quite understand what "Put open app ads." means?
Also, where did you call the callback for when something is wrong with the camera?

SweetD3v, 2022-08-08:
I'm getting too many crashes in this **onStartBind()** method. Anyone found any solution? @natario1 ?