natario1/CameraView #1175: setParameters failed and crash

leacode, 2022-06-05:
###  Describe the bug

I received an error online. 

- device: Blackview BV9800Pro
- os: Android Pie (v9)(Level 28)

data:text/text;charset=utf-8,
java.lang.RuntimeException: setParameters failed
	 at android.hardware.Camera.native_setParameters(Native Method)
	 at android.hardware.Camera.setParameters(Camera.java:2125)
	 at com.otaliastudios.cameraview.engine.Camera1Engine$11.run(Camera1Engine.java:884)
	 at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
	 at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:84)
	 at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:81)
	 at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152)
	 at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
	 at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147)
	 at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34)
	 at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
	 at android.os.Handler.handleCallback(Handler.java:873)
	 at android.os.Handler.dispatchMessage(Handler.java:99)
	 at android.os.Looper.loop(Looper.java:193)
	 at android.os.HandlerThread.run(HandlerThread.java:65)

### Expected behavior

It should not crash my app.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```
<com.otaliastudios.cameraview.CameraView
                    android:id="@+id/camera_preview"
                    android:layout_width="wrap_content"
                    android:layout_height="wrap_content"
                    android:keepScreenOn="true"
                    app:cameraAudio="off"
                    app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
                    app:cameraWhiteBalance="auto"
                    app:cameraAutoFocusResetDelay="0"
                    app:cameraEngine="camera2"
                    app:cameraGestureTap="autoFocus"
                    app:cameraMode="picture"
                    app:cameraPreview="glSurface"
                    app:cameraPreviewFrameRate="30"
                    app:cameraPreviewFrameRateExact="false"
                    app:cameraUseDeviceOrientation="false"
                     />
```

naymushin, 2022-08-04:
Hey @leacode! Hopefully my PR will get merged and this bug will go away. On a side note, you probably need to set the `app:cameraExperimental="true`" attrubute, otherwise the ` app:cameraEngine="camera2"` attribute has no effect and the library will fallback to camera1 API.

closed 2022-08-16 by natario1

leacode, 2022-08-17:
Thank you. I will try it.> Hey @leacode! Hopefully my PR will get merged and this bug will go away. On a side note, you probably need to set the `app:cameraExperimental="true`" attribute, otherwise the ` app:cameraEngine="camera2"` attribute has no effect and the library will fallback to camera1 API.

PR #1184 (2022-08-16): A couple of fixes and docs improvements
### Before you go
Unless this is a simple fix (typos, bugs with obvious solution), please open an issue first so that
we can discuss the best approach to address the problem. Without a reference issue and discussion, 
unfortunately, this PR will likely be ignored.

If the edited files were covered by tests, updated tests are required for merging. 
Please look into the tests folders and make sure you cover new code.

- Fixes: #1175 #1168
- Tests: no
- Docs updated: yes

### Solution
The exception in #1175 was simply wrapped in a try-catch block with a `Ca [... cut]