natario1/CameraView #1209: Getting NullPointerException with setDefaultBufferSize() in onStartBind()

SweetD3v, 2023-01-02:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.7.2
- Camera engine used: camera2
- Reproducible in official demo app: No. because it's crashlytics log & I haven't found anything to reproduce it yet.
- Device / Android version: 10, 11, 12 & some older devices
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): Yes

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera_view"
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            android:layout_centerVertical="true"
            android:layout_gravity="center"
            android:keepScreenOn="true"
            app:cameraAudio="on"
            app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
            app:cameraEngine="camera2"
            app:cameraExperimental="true"
            app:cameraFacing="back"
            app:cameraFlash="off"
            app:cameraGestureLongTap="none"
            app:cameraGesturePinch="zoom"
            app:cameraGestureScrollHorizontal="none"
            app:cameraGestureScrollVertical="exposureCorrection"
            app:cameraGrid="off"
            app:cameraHdr="on"
            app:cameraMode="picture"
            app:cameraPictureMetering="false"
            app:cameraPlaySounds="false"
            app:cameraPreview="glSurface"
            app:cameraPreviewFrameRate="30"
            app:cameraPreviewFrameRateExact="false"
            app:cameraRequestPermissions="true">
</com.otaliastudios.cameraview.CameraView>
```

### Logs
```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference
       at com.otaliastudios.cameraview.engine.Camera2Engine.onStartBind(Camera2Engine.java:506)
       at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:471)
       at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:467)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loopOnce(Looper.java:226)
       at android.os.Looper.loop(Looper.java:313)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```

yalamo, 2023-08-17:
Any one has idea for this issus?

savan15, 2023-09-29:
I have also face same issue and it is reported on firebase and google play console.

Any help will be appreciated.

PearKMA, 2023-12-22:
I found out in some cases onResume() calls open() but in onPause(), it does not close camera because camera is not actually open, then when onResume() is called again it causes this error. I tried calling open in onStart() and close it in onStop() and it worked fine, not sure if it is completely correct.
In my case, the error occurred when asking for permission and navigating to the app settings and returning immediately after.

gosala, 2024-12-13:
Getting same issue, Any solution